NOSALRO/robot_dart #13: Compilation / get_vector_mimic

jbmouret, 2018-11-25:
I tried to compile the master on mac (Mojave) with Dart 6.2.1. (according to waf configure)

I errors saying that get/set_vector_mimic was not declared.

I see in robot_control.hpp

```c++
#if DART_MAJOR_VERSION > 6 || (DART_MAJOR_VERSION == 6 && DART_MINOR_VERSION > 6)
            std::vector<size_t> _mimic_dofs;

            Eigen::VectorXd _get_vector_mimic(const Eigen::VectorXd& vec) const;
            Eigen::VectorXd _set_vector_mimic(const Eigen::VectorXd& vec) const;
#endif
```
... but no ifdef in the .cpp file. It makes sense that it cannot find it.

So... Should we require dart 6.6? Should I switch to dart 6.6?

costashatz, 2018-11-25:
> Should we require dart 6.6?

No. This was a bug. I fixed it in 1635ec1.

> Should I switch to dart 6.6?

I would advise you to use 6.7..

closed 2018-11-25 by costashatz

jbmouret, 2018-11-25:
Thanks. But the one we have in github.com/resibots/dart is 7.0

and using this 7.0.

```
[ 5/23] Compiling src/robot_dart/control/robot_control.cpp
../src/robot_dart/control/robot_control.cpp:37:72: error: no member named 'MIMIC' in 'dart::dynamics::Joint'
                if (joint->getActuatorType() == dart::dynamics::Joint::MIMIC)
```

Which version should we use?
                                                ~~~~~~~~~~~~~~~~~~~~~~~^

costashatz, 2018-11-25:
> But the one we have in github.com/resibots/dart is 7.0

This repo is just a fork of the original repo that I used to make a few fixes. We should not use this repo. Use the original one. Either master branch (v7.0) or release-6.7 branch. Both of them should work with robot_dart.

jbmouret, 2018-11-25:
OK. I will try with 6.7 because their master does not want to compile on my mac...