NOSALRO/robot_dart #14: -march=native causing segfault

Aneoshun, 2019-03-14:
Hi all, 

Robot_dart compiles, but when I run the examples (on a 18.04 + Dart 7.0 compiled from sources), I get a segfault. Removing -march=native from the wscript fixes the problem. 

Do you want a PR removing -march=native?

Best regards,

costashatz, 2019-03-14:
> Do you want a PR removing -march=native?

No! I PR that makes it optional would be nice though! I did the same for limbo.

Did you compile DART with -march=native enabled? I am saying it in the instructions: https://github.com/resibots/robot_dart/blob/master/docs/installation.md

Aneoshun, 2019-03-14:
arf... I read this too quickly apparently, and missed the SIMD activation. Works with this activated. Sorry for the spam.

costashatz, 2019-03-14:
I am leaving this open in order to add some catch around it and an optional argument to disable it..

costashatz, 2019-05-15:
Fixed in #17 ...

closed 2019-05-15 by costashatz