NOSALRO/robot_dart #18: Memory leak with the new off-screen cameras

Aneoshun, 2019-05-17:
When using the cameras in sferes (QD algorithms), the memory consumption rises linearly with the number of evaluation (while it should remain more or less constant at some point). 
To give an order of magnitude I reach 20GB of ram usage in less than 100 generations (pop size =200). 

I am investigating the issue, but if someone has some ideas or hints, they are welcome.

Aneoshun, 2019-05-20:
Apparently, the memory leak is not related to the camera_osr, but to the osg implementaiton in Dart (which is used by both the camera_osg and "graphics"), in particular in the WorldNode object. 
For instance, running the "arm" example with valgrind shows several issues. 
I continue my investigation.

Aneoshun, 2019-05-21:
(I am keeping note mainly for myself)
So I found two bugs in Dart causing memory leak.

First, in Dart::gui::osg::Viewer, the WorldNode(s) are not properly removed. 
calling `_osg_viewer->removeWorldNode(_osg_world_node);` in robot_dart::graphics, removes the issue.  I guess that it is because the worldnodes are stored in a std::map as pointers and not as osg::ref_ptr, which breaks the automatic garbage collection. 

Second, Dart::gui::osg::Viewer implements and uses a screenSaver callback (I am really wondering why this is not disabled by default). This callback is registered in the camera (in the viewer), but this callback also keeps a pointer of the camera. This creates a circular reference and breaks the garbage collection (both of these objects have always a pointer pointing to them). 
To solve this, the destructor of the Dart viewer should call: 
`getCamera()->setFinalDrawCallback(0);`

Thanks to these two fixes, I can run the following code with only 600KB in RAM instead of 30MB!! (yes, this code does nothing). 
```
int main(int argc, char **argv)
{
  for(size_t i=0; i<1000;i++){
    osg::ref_ptr<dart::gui::osg::Viewer> osg_viewer = new dart::gui::osg::Viewer;
    }
  return 0;
}
```

costashatz, 2019-05-21:
@Aneoshun great news!

Can you make a PR (to the master) for fixing this? Thanks a lot in advance!

Aneoshun, 2019-05-21:
@costashatz Most of the fixes will happen in Dart (not Robot_dart), I am continuing my investigation, in case I find other bugs, before reporting this to the Dart team. If anything needs to be fixed on Robot_dart, I will make a PR (from/to master)

costashatz, 2019-05-21:
> @costashatz Most of the fixes will happen in Dart (not Robot_dart), 

Yes of course.. But we can still be safe in our side (robot_dart) and fix them even if someone uses an older version of DART..

> I am continuing my investigation, in case I find other bugs, before reporting this to the Dart team.

Sure! Thanks for this in any case. Let us know if you find anything else..

PR #19 (2019-05-22): fix a leak from DART, this code should be compatible even after DART'…
fixes #18
closed 2019-05-22 by costashatz