NOSALRO/robot_dart #21: Off-screen cameras are not working in multi-threads setting

Aneoshun, 2019-05-22:
Now that the memory leak is solved, I see that the off-screen cameras are not really stable when running in parallel with TBB (in sferes). 

Observed symptomes:
1. Everything is working well if I use `eval::Eval`.
2. With `eval::Parallel`, the load is not properly distributed on the different cores. 
3. With `eval::Parallel`, the creation of the pbuffer fails after some point. 
4. With `eval::Parallel`, I have some random seg faults that happen after a couple of generations (but this might be caused by other elements in my code). 

I suspect that the creation of the pbuffer share some ressources or do some system calls that prevents everything to work well. Also, it seems that OSG is changing the processorAffinity, which can probably mess with TBB's assignment.

Aneoshun, 2019-05-22:
Using a more recent version of OSG (like the 3.6) offers access to the `setUseConfigureAffinity(bool flag)` function. Calling `_osg_viewer->setUseConfigureAffinity(false);` in the camera_osr.hpp enables the different threads to run on different cores, which should work better with TBB. However, it now crashes and valgrind reports that more than 500 threads have been created (both in OSG's singlethread and threadPerContext mods)...

Aneoshun, 2019-05-23:
So the problem seems to come mainly from the creation of the graphicContext (the pbuffer) `::osg::GraphicsContext::createGraphicsContext(traits.get())`

I have asked on the OSG forum for some advice about how to deal with this situation, it might take a while before getting some feedback. 

In the meanwhile, I have pushed (#22) a new way to managing the cameras. The idea is that several pbuffers are created before running the simulations (before starting sferes), then when a simulation is created, it can call a PbufferManager (as a global variable) to receive one of the previously created pbuffer and use it for its simulation. When the simulation is done, the pbuffer is returned to the PbufferManager. 

The PbufferManager takes care of giving a pbuffer available upon request. 

I tried it on some experiments with sferes and 64 threads and it works quite well (given that the creation of the context takes quite a long time, the speed improvement is REALLY significant. First I tought it was not generating the images...)

costashatz, 2019-07-08:
I guess this is fixed with #22 .. @Aneoshun let me know if this is not the case..

closed 2019-07-08 by costashatz