NOSALRO/robot_dart #23: Segmentation fault

costashatz, 2019-07-12:
[This line](https://github.com/resibots/robot_dart/blob/master/src/robot_dart/graphics/camera_osr.hpp#L93) gives me a segmentation fault in Ubuntu 18.04 with OSG 3.2.. Looking at the OSG code, I think that we do not need this line. @Aneoshun can you check if without this line the parallel implementation works fine?

costashatz, 2019-07-21:
@Aneoshun any news on this?

Aneoshun, 2019-07-22:
Hi @costashatz, That's weird as it does not create a segfault on my side with the same setup (18.04, OSG 3.2). However, this line is not really required. It's just to make sure that this pointer is reset, while this is normally done automatically. I tried my code with and without this line and I see no difference.

costashatz, 2019-07-22:
@Aneoshun Great! I am removing this line then from master.. Thanks!

closed 2019-07-22 by costashatz