NOSALRO/robot_dart #31: incompatible magnum_context

Aneoshun, 2019-11-01:
Hi @costashatz,

In order to test #30, I am migrating the virtual development environment of my lab to robot_dart with magnum. However, it does not go as smoothly as I expected: it seems that magnum is unhappy with the GL context. When I run any of the robot_dart examples (using magnum on the new remove_osg branch), I get: 
```
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Platform::Sdl2Application: can't get physical display DPI, falling back to no scaling: Couldn't get DPI
Platform::Sdl2Application::tryCreate(): cannot create core context: Could not create GL context (falling back to compatibility context)
Platform::Sdl2Application::tryCreate(): cannot create context: Could not create GL context
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Platform::Sdl2Application: can't get physical display DPI, falling back to no scaling: Couldn't get DPI
Platform::Sdl2Application::tryCreate(): cannot create core context: Could not create GL context: BadValue (integer parameter out of range for operation) (falling back to compatibility context)
Platform::Sdl2Application::tryCreate(): cannot create context: Could not create GL context: BadValue (integer parameter out of range for operation)
``` 
As stated, there is an issue with the GL context. the `magnum_context` example fails for some reasons that I don't understand: 
```
hader::compile(): compilation ofShader::compile(): compilation of  fragment shader fragment2 failed with the following message:
 shader 5:94(30): error: syntax error, unexpected NEW_IDENTIFIER, expecting '{'
2 failed with the following message:Assertion Magnum::GL::Shader::compile({vert, frag}) failed in ../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp on line 36
5:94(30): error: syntax error, unexpected NEW_IDENTIFIER, expecting '{'

Assertion Magnum::GL::Shader::compile({vert, frag}) failed in ../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp on line 36
Aborted (core dumped)
```

For information, here is the output of `glxinfo`
<details>

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
[... 825 lines cut ...]
0x176 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x177 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x178 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x179 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x17a 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x17b 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x17c 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x17d 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 32  0  0  0  0  0  0 0 None
0x17e 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x17f 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x180 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x181 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x182 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x183 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x184 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x185 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x186 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x187 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x188 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x189 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x18a 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None
0x18b 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None
0x18c 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 32  0  0  0  0  0  0 0 None
```
<\details>

costashatz, 2019-11-01:
@Aneoshun if I understood well, you have an OpenGL version 3.1. For the code that we have, we need an OpenGL version of at least 4.0. I think this is why the shader is not compiling (this is the only shader requiring version 4.0).

For the first errors (platform/context), I am not sure what is wrong.. I am trying to investigate it a bit more. Can you give me the details of your virtual development environment so that I can try to replicate it?

mosra, 2019-11-01:
If glxinfo is able to create a GL context, magnum *should* be able to do that too, at the very least the windowless applications (which they seem to do, seeing the second output). So there has to be something special with the SDL context -- @costashatz is there something special you're doing for GL context setup in the Sdl2Application?

> version 4.0

Mesa reports llvmpipe, which is a pure-software rendering, so things may run slower. Ideally (if this is a VM) it would be reporting SVGA3D, which does a pass-through from the host GPU. SVGA3D is quite buggy, but last time I looked at this, I added workarounds for the most pressing issues, so things should be good. Unfortunately in either case you'll have GL 3.3 at most, with a bunch of extensions from 4.0, but definitely not the full 4.0.

costashatz, 2019-11-01:
> @costashatz is there something special you're doing for GL context setup in the Sdl2Application?

Not really. Here's the code:

```cpp
Sdl2Application::Sdl2Application(int argc, char** argv, const dart::simulation::WorldPtr& world, size_t width, size_t height, const std::string& title, bool isShadowed)
                : BaseApplication(isShadowed), Magnum::Platform::Application({argc, argv}, Magnum::NoCreate), _speedMove(0.f), _speedStrafe(0.f)
            {
                /* Try 16x MSAA */
                Configuration conf;
                GLConfiguration glConf;
                conf.setTitle(title);
                conf.setSize({static_cast<int>(width), static_cast<int>(height)});
                conf.setWindowFlags(Configuration::WindowFlag::Resizable);
                glConf.setSampleCount(8);
                if (!tryCreate(conf, glConf))
                    create(conf, glConf.setSampleCount(0));
}
```

> Unfortunately in either case you'll have GL 3.3 at most, with a bunch of extensions from 4.0, but definitely not the full 4.0.

I can try to make the code GL 3.3 compatible..

mosra, 2019-11-01:
The GLX configs listed above don't have any MSAA support, so that's why it fails for the first time. However what is suspicious that it fails second time with a different error (`BadValue`). I double-checked that the MSAA config is reset back to [documented defaults](https://wiki.libsdl.org/SDL_GLattr) when specifying `0` so that *shouldn't* be an issue, but ... something is fishy and *somehow* the second `glConf.setSampleCount(0)` doesn't do what it is expected to do. Sigh.

@Aneoshun could you try [with the above changed](https://github.com/resibots/robot_dart/blob/d8a81f77dbf275dbdd72d76d8180da1a2e11a4e1/src/robot_dart/gui/magnum/sdl2_application.cpp#L14-L22) to just

```cpp
                /* Try 16x MSAA */
                Configuration conf;
                GLConfiguration glConf;
                conf.setTitle(title);
                conf.setSize({static_cast<int>(width), static_cast<int>(height)});
                conf.setWindowFlags(Configuration::WindowFlag::Resizable);
                create(conf, glConf);
```

costashatz, 2019-11-01:
@Aneoshun let us know if the fix from @mosra helps for the SDL2 problems..

I also made a small change in the `remove_osg` branch to allow compilation on OpenGL 3.3 (+1 feature from OpenGL 4.0, which your VM seems to have) and thus shader compilation should be able to work on your virtual machine..

Aneoshun, 2019-11-02:
Hi @costashatz and @mosra, 
Thanks a lot for your help. Unfortunately, the @costashatz 's new commit and the change suggested by @mosra does not solve the issue. 

I still get: 
```
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Platform::Sdl2Application: can't get physical display DPI, falling back to no scaling: Couldn't get DPI
Platform::Sdl2Application::tryCreate(): cannot create window: Couldn't find matching GLX visual
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Platform::Sdl2Application: can't get physical display DPI, falling back to no scaling: Couldn't get DPI
Platform::Sdl2Application::tryCreate(): cannot create core context: Could not create GL context: BadValue (integer parameter out of range for operation) (falling back to compatibility context)
Platform::Sdl2Application::tryCreate(): cannot create context: Could not create GL context: BadValue (integer parameter out of range for operation)
```
 and magnum_context gives:
```
Shader::compile(): compilation ofShader::compile(): compilation of  fragmentfragment shader shader Shader::compile(): compilation of 2 failed with the following message:
5:94(30): error: syntax error, unexpected NEW_IDENTIFIER, expecting '{'
2  fragment shader 2 failed with the following message:Assertion Magnum::GL::Shader::compile({vert, frag}) failed in ../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp on linefailed with the following message:
5:94(30): error: syntax error, unexpected NEW_IDENTIFIER, expecting '{'
Assertion Magnum::GL::Shader::compile({vert, frag}) failed in ../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp on line 36
5:94(30): error: syntax error, unexpected NEW_IDENTIFIER, expecting '{'
 36

Assertion Magnum::GL::Shader::compile({vert, frag}) failed in ../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp on line 36
Aborted (core dumped)
```
For your information, I get similar errors with Magnum examples. Please, let me know if there is anything else you would like me to try. 

I am not using a VM, but a singularity container. Singularity is like docker but designed to be used in research and in HPC environments. 
If you are interested in reproducing my environment, I have uploaded my container on the singularity cloud. 
After installing singularity (https://sylabs.io/guides/3.4/user-guide/quick_start.html), you can download a "sandbox" version of my environment with: `singularity build --sandbox airl_env library://airl_lab/default/airl_env:magnum`
(it will download the 2GB container, and create a writable copy of it)
Then, you can enter the environment with `singularity shell -w airl_env/`
Robot_dart is in `cd /git/robot_dart` and ready to compile with `./waf` (config already done).

costashatz, 2019-11-04:
@Aneoshun I am getting

`/usr/bin/ld: /workspace/lib/libMagnumSdl2Application.a(DpiScaling.cpp.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'`

I tried to uninstall and re-install magnum related repos, but I am still getting the same.. Any ideas?

Aneoshun, 2019-11-04:
I got the same for an unknown reason, but I solve this simply by adding `dl` as a needed lib in waf. (for instance, I added this in the waf_tool/magnum.py).

costashatz, 2019-11-04:
Now it compiled, but when I am running any example I get:

```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  113
  Current serial number in output stream:  112
```
Should I do something specific?

Aneoshun, 2019-11-04:
Ho yes, you should run our "visu_server" which starts an xvfb and a vnc server. 
`visu_server.sh &`

You will then be able to see the graphical environment in your browser at local_host:6080

costashatz, 2019-11-04:
```
Singularity airl_env:~/Workspaces/git/antoine/singularity/airl_env/git/robot_dart> Starting services for VISU_server
Starting Xinit (xdummy)
Starting X11vnc
Starting novnc
/workspace/bin/visu_server.sh: line 51: cd: /opt/noVNC: No such file or directory
Failed to start noVNC

Terminating x11vnc (22405)
Terminating xinit (22401)
pkill: killing pid 11911 failed: Operation not permitted
```

Aneoshun, 2019-11-04:
Sorry, that's my bad I did not included in this test environemnt (it's running by default on my dev server). Give me a moment, I will upload the complete environment...

Aneoshun, 2019-11-04:
I have uploaded the complete version of the enviornment. Same address as before, just make sure to delete your previous image (and maybe the cache in ~/.singularity)

costashatz, 2019-11-04:
> I have uploaded the complete version of the enviornment. Same address as before, just make sure to delete your previous image (and maybe the cache in ~/.singularity)

Can you give me the singularity definition file that you used to create the environment? That way I can play a bit around with it easierly (e.g., install some packages that I can trace opengl errors etc)..

Aneoshun, 2019-11-04:
Here you go: 
[airl_env_base.txt](https://github.com/resibots/robot_dart/files/3803476/airl_env_base.txt)
(I have renamed it as `.txt`, just to avoid github to complain, the normal extension is `.def`)

costashatz, 2019-11-04:
Still not able to get something to run:

```
Singularity airl_env:~/Workspaces/git/antoine/singularity/airl_env/git/robot_dart> Starting services for VISU_server
Starting Xinit (xdummy)
Starting X11vnc
Starting novnc

Singularity airl_env:~/Workspaces/git/antoine/singularity/airl_env/git/robot_dart> ./build/hexapod
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  113
  Current serial number in output stream:  112
```

Here's the output screen from localhost:6080:

[image]

Is there another way of running a visual app? Or any ideas to fix this?

Aneoshun, 2019-11-04:
hum.. this is strange. It usually works pretty easily. You should have the logs of Xinit, X11vnc and novnc in /opt

If you the container on a machine with a graphical environment and not inside ssh, you can unset Display, and I guess it should directly forward use your Xserver (I haven't played much in this context).

Aneoshun, 2019-11-04:
Also, sometimes you need to run `xhost + local:` to allow connections

costashatz, 2019-11-04:
I had to run the following `singularity shell --nv --writable-tmpfs airl_env` to make it use my native screen..

Nevertheless, I was not able to reproduce your behavior. In my case, it runs smoothly and nicely! I only get this warning for Sdl2Application: `Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI`.

Any ideas on what to do to re-produce the behavior?

Aneoshun, 2019-11-04:
I guess the main difference is the display environment. So to reproduce the behavior we need to fix the visu_server.sh

Could you give me the content of the following files in the container?
`/opt/xinit.log`
`/opt/x11vnc.log`

My current hypothesis is that the docker image (from nvidia) that I used to build my environment has been recently changed and they removed some packages (for instance, `less` ). One of the packages required by xinit is `xterm` that should be installed. Maybe you can try a `sudo apt-get update & sudo apt-get install xterm` (you need to run the container as root or with the `--fakeroot` option).

costashatz, 2019-11-04:
> I guess the main difference is the display environment. So to reproduce the behavior we need to fix the visu_server.sh

How can I install the `visu_server.sh` (not downloading your image, but when building mine)?
You have these lines in the definition file:

```
mv /workspace/tmp/etc/X11/* /etc/X11/
mv /workspace/tmp/etc/skel/.xinitrc /etc/skel/
mv /workspace/tmp/etc/xdg/openbox/* /etc/xdg/openbox/
```

Aneoshun, 2019-11-04:
You need the content of this zipfile next to the definition file when you build: 
[resources.zip](https://github.com/resibots/robot_dart/files/3804732/resources.zip)

costashatz, 2019-11-08:
@Aneoshun please try again with the latest `remove_osg` branch. Make sure that you adapt your installation scripts: you now need glfw3 (glfw3-dev I guess for ubuntu) and also you need to change the cmake configuration of Magnum (see the update installation.md in `remove_osg`). You also need the latest magnum (to apply some fixes that I made).

Hopefully now it works in your container. Using these changes, I made it work in my local singularity container.

Aneoshun, 2019-11-08:
It works!
I just get those two warnings: 
```
X11: RandR gamma ramp support seems broken
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
```
I don't know if it is important/solvable, but in any case it works. 

I also, I have 2 shadows in the meshes and hexapod examples (the only two I tested), it this normal?

[image]

costashatz, 2019-11-08:
> It works!

Great!

> I just get those two warnings:

I was getting similar warning! The first one I have no idea what it is. The second one just says that there's no real DPI available (which makes sense) and it harmless.

> I also, I have 2 shadows in the meshes and hexapod examples (the only two I tested), it this normal?

Yes. By default there are two directional lights. Is it confusing? Should I just use one light as default? With one light some parts of the objects are too dark because of the hitting angle; that's why I put 2 lights and then you get 2 shadows. What do you think?

In any case, users can remove the lights and put any light that they want.

Aneoshun, 2019-11-08:
I would say that it is a bit confusing because they are both of similar intensity. So, I was wondering if it was a bug. But that's not a big deal.

costashatz, 2019-11-08:
> I would say that it is a bit confusing because they are both of similar intensity.

I can put only one light and some small ambient values to not get too dark parts.

Anyway, closing this issue.. Thanks for the feedback..

closed 2019-11-08 by costashatz