NOSALRO/robot_dart #37: Clean/Improve Graphics

costashatz, 2020-05-10:
The Magnum GUI is working well, but there remains to be a lot of cleaning and more features. In particular:

- Clean up the code related to graphics/Magnum
- Go through the OpenGL stuff and investigate if we can be better/faster
- Provide all the parameters to the user (e.g., shadow map sizes, light features, etc.)

costashatz, 2020-05-11:
One idea to (possibly) make the GUI faster:

- call draw(), let the gpu work on that
- simulate physics while the gpu works
- only then call swapbuffers

Instead of the current:
- simulate physics (takes CPU time)
- draw() (takes GPU time)
- swapbuffers (CPU idle, waits until GPU is done)

To test if it is actually going to help..

costashatz, 2020-05-11:
Other ideas to improve rendering times (again credits to @mosra):

- Instancing, e.g. see [here](https://learnopengl.com/Advanced-OpenGL/Instancing)
- When constructing the Magnum GUI, the user could request the types of lights and max lights per type; this way we do not need to allocate before-hand all the textures (saves quite a lot of GPU memory)

One more thing to check is the way I am plotting transparent objects (I have a feeling that I am wasting too much time there, even if we do not have any transparency).

costashatz, 2020-05-13:
A first cleaning was done with #47... Another thing to investigate is the defines for the plugins; it should not be needed for newer versions of Magnum.

jbmouret, 2020-05-18:
I think the graphics do not need to be faster (graphics is not slowing us down, I think). But simpler is better for long-term maintenance. 

Also, we need maximum performance in no graphics mode. We usually do not care much in graphics mode (in the worst case, we record a video and play it at the right framerate).

Aneoshun, 2020-05-18:
I agree that graphics should remain simple for long-term maintenance. However, we are more and more using cameras during the training, and more and more in video streams. So, faster graphics without overwhelming complexity would be welcome. 

(And I agree, max perf in no graphics mode is priority).

costashatz, 2020-05-18:
> I agree that graphics should remain simple for long-term maintenance. However, we are more and more using cameras during the training, and more and more in video streams. So, faster graphics without overwhelming complexity would be welcome.

I tend to agree with Antoine. I am planning to use the cameras quite a lot in the coming future and thus faster graphics would make my life easier. Of course, faster does not necessarily means more complex.

> Also, we need maximum performance in no graphics mode.

I think we do not really add any serious overhead over plain DART, do we?

costashatz, 2020-06-18:
Implement Cascaded Shadow Maps for Directional Lights (see [here](https://doc.magnum.graphics/magnum/examples-shadows.html) and [here](https://docs.microsoft.com/en-us/windows/win32/dxtecharts/cascaded-shadow-maps#:~:text=Cascaded%20shadow%20maps%20(CSMs)%20are,errors%20with%20shadowing%3A%20perspective%20aliasing.&text=describes%20the%20two%20most%20common,variance%20shadow%20maps%20(VSMs)%3B)).