NOSALRO/robot_dart #38: Last master does not compile anymore

jbmouret, 2020-05-10:
Hey

I pulled the last master (after your merge) and it does not compile anymore on my mac.

```
../src/robot_dart/gui/magnum/gs/phong_multi_light.cpp:17:21: error: static_assert failed due to requirement 'Attribute<2, Magnum::Math::Vector3<float> >::Location == 5' "Shader Normal Attribute Location Error!"
                    static_assert(Normal::Location == NORMAL_ATTRIBUTE_LOCATION, "Shader Normal Attribute Location Error!");
```

=> Maybe I need to update Magnum too (you see, this is why it would be nice to not have to maintain the graphics part...)? any idea about what you changed recently?

jbmouret, 2020-05-10:
This comes from this commit: https://github.com/resibots/robot_dart/commit/b5fab1fbd11fbdd7338733fc8965d6c760aafc20

So... I guess I will try to recompile Magnum again.

costashatz, 2020-05-10:
You should pull the latest Magnum yes.. I am in the process of making all these to work independently of the versions of the graphics..

jbmouret, 2020-05-10:
OK. The update via brew did not work, but I managed to re-compile everything by hand.

closed 2020-05-10 by jbmouret

costashatz, 2020-05-10:
> OK. The update via brew did not work, but I managed to re-compile everything by hand.

As the installation instructions (for the moment) say, you need to compile the latest master branch of Magnum.

I will fix these issues with the refactoring...

jbmouret, 2020-05-10:
Sure, but "the last magnum" does not mean "recompile the last master of magnum + corrade + etc. every morning"...

Don't Magnum has some release numbers so that we can use a specific version?

costashatz, 2020-05-10:
> Don't Magnum has some release numbers so that we can use a specific version?

They have.. This is what I will fix. Take into account Magnum releases. They have a release every ~4 months.. I am trying to find the best way to check for their version in code and act accordingly.