NOSALRO/robot_dart #41: Add API for drawing helper text/2D visualizations

costashatz, 2020-05-10:
We should add helper functions for the user to easily add text and 2D visualizations on the screen. Here's a list:

- Text that can be changed at every time-step
- 2D rectangles with colors/fillings
- ...

@jbmouret @dalinel @PedroDesRobots let me know if you have something to add..

costashatz, 2020-05-11:
@jbmouret @dalinel @PedroDesRobots would it make sense to have a more complete UI support (e.g., like [this](https://magnum.graphics/showcase/ui-gallery/))? I think that we would need mainly only text, so it'll be too much to incorporate properly a UI (Magnum has a built-in, but we would have to expose the generic API somehow).. The text-only is quite easy to do.. Let me know what you think..

jbmouret, 2020-05-11:
I think some text is enough but if we get something nicer for free, why not.

costashatz, 2020-05-11:
> I think some text is enough

I think I will do that so that we can have something quickly.

> if we get something nicer for free, why not.

The built-in Magnum UI does not come for free, it just comes easier than ImGui (and less dependencies). Why would we need a full UI though? Do we want to expose some user interaction? Although we could to be honest :P

jbmouret, 2020-05-12:
If there are additional dependencies, I think we should stay with the text. If this is already a component of Magnum that we simply have to activate, then we could use it as it would allow more in the future (e.g., press a button to have a random force that pushes the robot).

jbmouret, 2020-07-17:
Any news of this? it would be nice to have a minimal 'dashboard' about the simulation (at least the time like Gazebo). Most things are robot-specific but we could provide a Talos-dashboard, for example, with the values of the sensors (IMU, force sensors of the feet, for instance).

costashatz, 2020-07-17:
> Any news of this? it would be nice to have a minimal 'dashboard' about the simulation (at least the time like Gazebo).

No news. The easiest way would be to use the (somewhat limited, but easy to use) `Magnum-UI` which is another library but can be seen as a Magnum component (and thus not really a different dependency). Let me see if I can do this before the end of July and get back to you..

> Most things are robot-specific but we could provide a Talos-dashboard, for example, with the values of the sensors (IMU, force sensors of the feet, for instance).

These specific things should go to your Talos code. We should however provide a relatively simple API to add those things. But for this, I should probably first finish the Sensor API (see #79).

costashatz, 2020-09-24:
This is partly fixed by #98, but leaving this open for adding the rest of the visualizations.

jbmouret, 2020-09-25:
Testing the current API:
- we need many more characters (like [, /. ...).
- we should define placeholders and the user can put text there. I propose :
  - the top left screen
  - a status line at the bottom
=> this should cover most of the use cases for simple display / debugging

costashatz, 2020-10-10:
I think this is fixed for now with #101 ..

PR #98 (2020-09-24): Text 2D Visualizations
This PR enables us to add 2D text visualizations (with different colors, locations, sizes, rotations). It partly fixes #41...
closed 2020-10-10 by costashatz