NOSALRO/robot_dart #44: Add proper documentation

costashatz, 2020-05-10:
We should have a better documentation with automatic API docs and possibly a website?

costashatz, 2020-05-10:
I see two options for creating a website:

- https://readthedocs.org/
- https://blog.getpelican.com/

I guess for the API documentation only [doxygen](http://www.doxygen.nl/) is there?

@jbmouret @dalinel @PedroDesRobots can someone start preparing this? I liked [limbo](http://www.resibots.eu/limbo/index.html)'s approach and design, but pelican seems to be very easy to use also (I remember that we had many issues with sphinx)..

jbmouret, 2020-05-10:
Sphinx + doxygen is not trivial to setup...

What does Pelican? it replaces sphinx?

costashatz, 2020-05-10:
> What does Pelican? it replaces sphinx?

Yes. I am not super familiar with it, but it was certainly a bit easier to use that what I remember with limbo's website.

Magnum's website is using it and seems to be quite good. Here's a few links:

- [m.css](https://mcss.mosra.cz/)
- [m.css with Pelican](https://mcss.mosra.cz/themes/pelican/)
- [Pelican Magnum](https://blog.magnum.graphics/meta/new-website-is-here/)
- [Pelican/m.css with Doxygen](https://mcss.mosra.cz/documentation/doxygen/)
- [Pelican/m.css for API docs of python files](https://mcss.mosra.cz/documentation/python/)

I haven't used all these features, but I did manage at some point to create a dummy website with Pelican and m.css in less than 10 minutes time.

I will try to gather more information about this..

jbmouret, 2020-05-11:
If we can avoid doxygen + sphinx, I am all for it. I am not sure I will ever be able to make it work again :)

What I wanted with Limbo, however, was to be able to call doxygen results in a text, instead of having these huge doxygen-generated documentation (it's like, doxygen results become a library, it's not independent).

We did things like  this:
https://github.com/resibots/limbo/blob/master/docs/api.rst 

Where `.. doxygenclass::  limbo::model::MultiGP`` is replaced by the link + documentation of this.

jbmouret, 2020-05-12:
Indeed, Magnum + doxygen does not seem bad: https://doc.magnum.graphics/magnum/getting-started.html

costashatz, 2020-05-12:
Do not forget to check the super cool and fast search: https://doc.magnum.graphics/magnum/getting-started.html#search

PR #145 (2024-05-01): Documentation
This PR adds documentation and website for robot_dart. This fixes #44...
closed 2024-05-01 by costashatz