NOSALRO/robot_dart #46: compilation error magnum and fcl

dalinel, 2020-05-11:
Hello,
I have the following compilation error : 
```
[37/51] Compiling src/robot_dart/gui/magnum/resources/resources.conf
In file included from ../../../rf_ws/devel/include/dart/collision/fcl/BackwardCompatibility.hpp:87:0,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLTypes.hpp:37,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLCollisionDetector.hpp:38,
                 from ../src/examples/meshes.cpp:7:
/usr/include/fcl/octree.h: In member function ‘fcl::OcTree::OcTreeNode* fcl::OcTree::getNodeChild(fcl::OcTree::OcTreeNode*, unsigned int)’:
/usr/include/fcl/octree.h:198:18: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
     return tree->getNodeChild(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘const OcTreeNode* fcl::OcTree::getNodeChild(const OcTreeNode*, unsigned int) const’:
/usr/include/fcl/octree.h:208:18: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
     return tree->getNodeChild(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘bool fcl::OcTree::nodeChildExists(const OcTreeNode*, unsigned int) const’:
/usr/include/fcl/octree.h:218:18: error: ‘const class octomap::OcTree’ has no member named ‘nodeChildExists’
     return tree->nodeChildExists(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘bool fcl::OcTree::nodeHasChildren(const OcTreeNode*) const’:
/usr/include/fcl/octree.h:228:18: error: ‘const class octomap::OcTree’ has no member named ‘nodeHasChildren’
     return tree->nodeHasChildren(node);
                  ^

In file included from ../../../rf_ws/devel/include/dart/collision/fcl/BackwardCompatibility.hpp:87:0,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLTypes.hpp:37,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLCollisionDetector.hpp:38,
                 from ../src/examples/meshes.cpp:7:
/usr/include/fcl/octree.h: In member function ‘fcl::OcTree::OcTreeNode* fcl::OcTree::getNodeChild(fcl::OcTree::OcTreeNode*, unsigned int)’:
/usr/include/fcl/octree.h:198:18: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
     return tree->getNodeChild(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘const OcTreeNode* fcl::OcTree::getNodeChild(const OcTreeNode*, unsigned int) const’:
/usr/include/fcl/octree.h:208:18: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
     return tree->getNodeChild(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘bool fcl::OcTree::nodeChildExists(const OcTreeNode*, unsigned int) const’:
/usr/include/fcl/octree.h:218:18: error: ‘const class octomap::OcTree’ has no member named ‘nodeChildExists’
     return tree->nodeChildExists(node, childIdx);
                  ^
/usr/include/fcl/octree.h: In member function ‘bool fcl::OcTree::nodeHasChildren(const OcTreeNode*) const’:
/usr/include/fcl/octree.h:228:18: error: ‘const class octomap::OcTree’ has no member named ‘nodeHasChildren’
     return tree->nodeHasChildren(node);

```
It is certainly due to a fcl versioning issue

On my computer I have
```
libfcl-0.5-dev/now 0.5.0-3osrf1~xenial3 amd64 [installed,local]
  Flexible Collision Library - development files

liboctomap-dev/xenial,now 1.6.8+dfsg-2.1 amd64 [installed]
  Octomap library development files

ros-kinetic-octomap/now 1.8.1-0xenial-20180809-134711-0800 amd64 [installed,local]
  The OctoMap library implements a 3D occupancy grid mapping approach, providing data structures and mapping algorithms in C++.

ros-kinetic-octomap-msgs/now 0.3.3-0xenial-20181107-010952-0800 amd64 [installed,local]
  This package provides messages and serializations / conversion for the OctoMap library.

ros-kinetic-octomap-ros/now 0.4.0-0xenial-20181117-015315-0800 amd64 [installed,local]
  octomap_ros provides conversion functions between ROS and OctoMap's native types.
```

I cannot do as described in the robot_dart installation process : 
``` sudo apt-get install libfcl-dev```
 Because it would uninstall a lot of ros packages

costashatz, 2020-05-11:
> I cannot do as described in the robot_dart installation process :
> ` sudo apt-get install libfcl-dev`
> Because it would uninstall a lot of ros packages

This is a bit weird. What version of ROS did you install? Why do you need a custom `fcl-dev` for ROS?

dalinel, 2020-05-11:
I am working with ros-kinetic , and on a specific environment dedicated to talos

A lot of talos specific packages would be remove
And also ros-kinetic-moveit related packages

costashatz, 2020-05-11:
> I am working with ros-kinetic , and on a specific environment dedicated to talos
> 
> A lot of talos specific packages would be remove
> And also ros-kinetic-moveit related packages

I see.

Still this is weird because DART is supposed to work with any version of fcl.

> libfcl-0.5-dev/now 0.5.0-3

I have libfcl-0.5.0-5.

Can you try unsourcing your ROS environment and doing:

- ./waf distclean
- ./waf configure
- ./waf

?

dalinel, 2020-05-11:
Ok so actually the problem is with octomap
I can install dart using libfcl-0.5-dev and the embedded ros-kinetic liboctomap.so.1.8.1

When it is like this I now get this error with robot_dart 
```
[30/51] Compiling src/examples/cameras.cpp
In file included from ../../dart_install/include/dart/collision/fcl/BackwardCompatibility.hpp:87:0,
                 from ../../dart_install/include/dart/collision/fcl/FCLTypes.hpp:37,
                 from ../../dart_install/include/dart/collision/fcl/FCLCollisionDetector.hpp:38,
                 from ../src/examples/meshes.cpp:7:
/usr/include/fcl/octree.h:50:29: fatal error: octomap/octomap.h: No such file or directory
compilation terminated.
```

If I install 
```
sudo apt-get install liboctomap-dev
```
It will install the liboctomap1.6.8 version 
If I use this and not the liboctomap1.8.1 I will have the same following error when installing dart 
```
In file included from ../../../rf_ws/devel/include/dart/collision/fcl/BackwardCompatibility.hpp:87:0,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLTypes.hpp:37,
                 from ../../../rf_ws/devel/include/dart/collision/fcl/FCLCollisionDetector.hpp:38,
                 from ../src/examples/meshes.cpp:7:
/usr/include/fcl/octree.h: In member function ‘fcl::OcTree::OcTreeNode* fcl::OcTree::getNodeChild(fcl::OcTree::OcTreeNode*, unsigned int)’:
/usr/include/fcl/octree.h:198:18: error: ‘const class octomap::OcTree’ has no member named ‘getNodeChild’
     return tree->getNodeChild(node, childIdx);
```

So in the end, the problem is that robot_dart cannot find 
/opt/ros/kinetic/include/octomap/octomap.h

costashatz, 2020-05-11:
> It will install the liboctomap1.6.8 version
> If I use this and not the liboctomap1.8.1 I will have the same following error when installing dart

First of all, this is weird. It never happened to me. DART should work with the latest octomap (v1.6.8 is not even the latest). I am using DART with Octomap v1.9.5. The FCL version with octomap are getting mixed. You should install the FCL version that corresponds to the octomap version that you want.

What version of DART are you using? Which branch?

Can you give me a docker or singularity container so that I can try with the same system?

> I can install dart using libfcl-0.5-dev and the embedded ros-kinetic liboctomap.so.1.8.1
> 
> When it is like this I now get this error with robot_dart

Of course, because you are installing an FCL version that has different header files than the ones that octomap has.

I will check to see if I can make a fix for this even if there are mismatched versions installed. Although I doubt that it is possible.

Can you send me a docker/singularity container with the system that you have (more or less) so that I can try to see what is the issue and how to fix this?

dalinel, 2020-05-11:
I am on the master branch of DART, I have also tried with v6.9.2

Here is a link to the docker with the same error : 
[https://gitlab.inria.fr/locolearn/public/docker_dart_wrapper/container_registry](url)
The docker is this one 
[registry.gitlab.inria.fr/locolearn/public/docker_dart_wrapper/robotdart](url)
You can modify this script to start the docker
[https://gitlab.inria.fr/locolearn/public/docker_dart_wrapper/-/blob/master/run_docker.sh](url)

 libfcl-0.5-dev and the embedded ros-kinetic liboctomap.so.1.8.1 are used
Every robot_dart dependency is installed in the /home/user/install directory

I configure robot_dart this way:
```
./waf configure --prefix=/home/user/install --dart=/home/user/install --corrade=/home/user/install --magnum_install_dir=/home/user/install --magnum_integration_install_dir=/home/user/install --magnum_plugins_install_dir=/home/user/install

```

dalinel, 2020-05-12:
Yes perfect this fixes the issue

PR #49 (2020-05-12): Fix DART detection (with octomap)
This PR attempts to fix #46.. It is not perfect, but I think this is OK for now.

What the authors of DART did is insane! They have an optional dependency that they fuse inside the main dart library! Not a different one!

In any case, @dalinel  can you check that it solves your issue (#46)? I checked in the docker image that you sent me and it compiles (beware of `LD_LIBRARY_PATH`).
closed 2020-05-12 by costashatz