NOSALRO/robot_dart #50: Releases and changleog

costashatz, 2020-05-12:
I think we should start organizing for releases and a changelog.

costashatz, 2020-05-15:
I propose a release scheme as follows:

- We do a release only when we really have something to release or when minor patches stay in master for more than 2-3 months
- Release versioning: major.minor.patch; we start from v1.0.0 once we finish this restructure
    - we increment the major number when we have API breakage (even if we keep backward compatibility)
    - we increment the minor number when we have serious bug-fixes or important new features
    - we increment the patch number for everything else (e.g., fix of corner-cases, minor new features etc.)
- Changelog
    - Every PR to be merged should add changelog
    - We begin keeping track of changes after v1.0.0 (this iteration)

@jbmouret @dalinel @PedroDesRobots @Aneoshun  Let me know if you agree or have anything to add.

Aneoshun, 2020-05-15:
Thanks Konstantinos, 

Sounds good to me.

costashatz, 2024-05-03:
Closed by #145

closed 2024-05-03 by costashatz