NOSALRO/robot_dart #52: Install the robot URDF in some 'path'?

jbmouret, 2020-05-12:
Should we install the URDF files in something like $PREFIX/robot_dart/robots/ ?

Because for now we need to copy/link them in each project.

jbmouret, 2020-05-18:
How do we include the URDF?
- as submodules?
- as files directly (with meshes, etc.)

And where do we install them?

I feel we need to have well-curated URDF because 'random urdf' usually do not work well. We also need to tune the collision shapes, etc. for maximum performance.

costashatz, 2020-05-18:
> I feel we need to have well-curated URDF

I agree with you here.

> And where do we install them?

Let's first finish the main restructuring of the library and then we should discuss on how to add example robots in robot_dart (along with their URDFs).

jbmouret, 2020-06-02:
We are almost there :) Thinking about it, I would suggest to create classes for each robot (Talos, Iiwa, etc.) because this would make it easier to define sensors and accessors with intuitive names.

Each class would use a default URDF, which would be taken from either a 'res' directory if it exists, from a ROBOT_DART_PATH env variable, or from `$PREFIX/share/robot_dart/robots` (installed with robot_dart). Of course, the user could use a different URDF with the same robot class (in that case, the user provides the path). $PREFIX should be taken from the compilation line.

The `res` directory is to make sure that the developers of robot_dart are using the urdf of the source tree and not the urdf that are installed on the system. The users should use the system files. The env. variable is to make it possible to have several installations of robot_dart files / switch etc.

closed 2020-08-08 by costashatz