NOSALRO/robot_dart #53: Save video

jbmouret, 2020-05-12:
We need to save a video of an execution (useful for papers, for replays, etc.).

PedroDesRobots, 2020-05-13:
We did it with OSG : 
```
void bridge::dart_robot_loader::dartRecord(const std::string &directory,
                                           const std::string &prefix,
                                           bool restart,
                                           std::size_t digits){

  boost::lock_guard<boost::mutex> lock(simMutex_);
  osg_viewer_->record(directory,prefix,restart,digits);
}
```
But I don't find yet a `record` function inside Magnum's API.

costashatz, 2020-05-13:
> We did it with OSG :

This is recording images per time-step and saving them in a directory, right?
This is already possible with Magnum. You can call the `image()` function (see [here](https://github.com/resibots/robot_dart/blob/master/src/examples/cameras.cpp#L76)) either inside a `Descritptor` or after N steps if you do the steps by hand and save them whereever you want using the (save_png_image)(https://github.com/resibots/robot_dart/blob/master/src/examples/cameras.cpp#L81) function.

What we were discussing with @jbmouret was to be able to directly save to a video.

PedroDesRobots, 2020-05-13:
> This is recording images per time-step and saving them in a directory, right?

Yes. Then we were transforming after with `ffmpeg`.

 But I understand that you want to directly save to a video.

costashatz, 2020-05-13:
> Yes. Then we were transforming after with `ffmpeg`.

This is already possible with the current API.

> But I understand that you want to directly save to a video.

We will see if we can do this somehow automatically.

jbmouret, 2020-05-13:
Yes, like in pyhexapod, we can stream the image directly to ffmpeg to get a mp4 video directly from the images. The hardest part is to get the picture format right.

jbmouret, 2020-05-18:
see #58

costashatz, 2020-05-20:
Fixed by #58.

closed 2020-05-20 by costashatz