NOSALRO/robot_dart #63: Simplify APIs

jbmouret, 2020-05-18:
I feel there is some boilerplate code that can be deleted by exposing the objects directly. For instance, in graphics.hpp
```c++

 void set_speed(const Magnum::Vector2& speed) { _magnum_app->camera().set_speed(speed); }
                void set_near_plane(double near_plane) { _magnum_app->camera().set_near_plane(near_plane); }
                void set_far_plane(double far_plane) { _magnum_app->camera().set_far_plane(far_plane); }
                void set_fov(double fov) { _magnum_app->camera().set_fov(fov); }
                void set_camera_params(double near_plane, double far_plane, double fov, size_t width, size_t height) { _magnum_app->camera().set_camera_params(near_plane, far_plane, fov, width, height); }

                Magnum::Vector2 speed() const { return _magnum_app->camera().speed(); }
                double near_plane() const { return _magnum_app->camera().near_plane(); }
                double far_plane() const { return _magnum_app->camera().far_plane(); }
                double fov() const { return _magnum_app->camera().fov(); }

```
=> this can easily be replaced by a camera() accessor (then we can call fov, far_plane(), etc on the camera)

jbmouret, 2020-05-19:
- What is the convention: should we return a `gs::Camera&` or a gs::Camera*` ? I see both in the code (for instance, a pointer to the magnum_app but a reference to the lights)

- should we do the same for magnum_app?

```
  void set_recording(bool recording, bool recording_depth = false) { _magnum_app->record(recording, recording_depth); }
                bool recording() { return _magnum_app->recording(); }
                bool recording_depth() { return _magnum_app->recording_depth(); }

                bool shadowed() const { return _magnum_app->shadowed(); }
                bool transparent_shadows() const { return _magnum_app->transparent_shadows(); }
                void enable_shadows(bool enable = true, bool transparent = true) { _magnum_app->enable_shadows(enable, transparent); }
```

... or you want to hide this app?

costashatz, 2020-05-19:
> * What is the convention: should we return a `gs::Camera&` or a `gs::Camera*` ? I see both in the code (for instance, a pointer to the magnum_app but a reference to the lights)

This is true. We should only keep one of them. I would say that I prefer references. A small care should be taken with references and the python bindings, but I can be checking this.

> * should we do the same for magnum_app?

I do not remember the reason why I decided to re-write those functions. I mean I am giving the magnum_app pointer so one could use this. I think that initially I wanted to hide `magnum_app`, but then I needed it. Let me check if we can hide it. I think that we should hide it if possible, because this can change and it should not affect the public API.

jbmouret, 2020-05-19:
Yes, I agree that the Magnum_app is not as intuitive as accessing a camera or a light.

At the same time, we could merge Graphics and BaseApplication quite easily.

closed 2020-05-20 by costashatz