NOSALRO/robot_dart #64: Create packages

jbmouret, 2020-05-18:
We should provide packages. The obvious ones are:
- Ubuntu via a PPA
- brew for OSX
- pip for Python (which will work for both ubuntu and mac).

costashatz, 2020-05-19:
> * Ubuntu via a PPA

I am on to this.

> * brew for OSX

Do you know how to do this? I guess this is like AUR of arch-linux? Btw, I can create an arch-linux package. It's super easy to do.

> * pip for Python (which will work for both ubuntu and mac).

pip should also be easy. I think we just need a `setup.py`? Or do you mean to push it on the repos?

jbmouret, 2020-05-19:
I will take care of brew (this is a ruby script that compiles everything) and there is a "ppa-style" way of adding repos (tap)

Pip is probably the most challenging because this means handling all the c++ dependencies. I think this is what usually happens with 'native' libraries.

There is also an alternative way of thinking which is that if we have pip that compiles the C++, maybe we only need this one.

costashatz, 2021-10-02:
I am removing the `1.0.0` milestone and we will handle this in later release.