NOSALRO/robot_dart #82: All examples core dump when running

danachee, 2020-06-06:
This is built on a Ubuntu 18.04 docker following all of the directions for adding requirements.  Each example dies when trying to parse and build from its urdf file.  I've tried using DART master as well, which doesn't help.  Any ideas on what I could have missed that's causing this issue?

Compiled:
~~~
g++ -g -I/usr/include/eigen3 -o /tmp/hex src/examples/hexapod.cpp -l RobotDARTSimu -ldart
~~~

Run (gdb /tmp/hex):
~~~
Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x3a1) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x3a1) at malloc.c:3103
#1  0x00007fa6fc740f5b in dart::common::EmbeddedStateAndPropertiesAspect<dart::dynamics::GenericJoint<dart::math::SE3Space>, dart::dynamics::detail::GenericJointState<dart::math::SE3Space>, dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SE3Space> >* dart::common::SpecializedForAspect<dart::common::EmbeddedStateAndPropertiesAspect<dart::dynamics::GenericJoint<dart::math::SE3Space>, dart::dynamics::detail::GenericJointState<dart::math::SE3Space>, dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SE3Space> > >::_createAspect<dart::dynamics::FreeJoint::Properties const&>(dart::common::SpecializedForAspect<dart::common::EmbeddedStateAndPropertiesAspect<dart::dynamics::GenericJoint<dart::math::SE3Space>, dart::dynamics::detail::GenericJointState<dart::math::SE3Space>, dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SE3Space> > >::type<dart::common::EmbeddedStateAndPropertiesAspect<dart::dynamics::GenericJoint<dart::math::SE3Space>, dart::dynamics::detail::GenericJointState<dart::math::SE3Space>, dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SE3Space> > >, dart::dynamics::FreeJoint::Properties const&) [clone .isra.593] () from /usr/local/lib/libdart.so.6.9
#2  0x00007fa6fc741f27 in dart::dynamics::FreeJoint::FreeJoint(dart::dynamics::FreeJoint::Properties const&) () from /usr/local/lib/libdart.so.6.9
#3  0x00007fa6fcc70ef5 in dart::dynamics::Skeleton::createJointAndBodyNodePair<dart::dynamics::FreeJoint, dart::dynamics::BodyNode> (this=0x55ca0103a9c0, 
    _parent=0x0, _jointProperties=..., _bodyProperties=...)
    at /usr/local/include/dart/dynamics/detail/Skeleton.hpp:82
---Type <return> to continue, or q <return> to quit---
#4  0x00007fa6fb596b0b in dart::utils::DartLoader::modelInterfaceToSkeleton(urdf::ModelInterface const*, dart::common::Uri const&, std::shared_ptr<dart::common::ResourceRetriever> const&) () from /usr/local/lib/libdart-utils-urdf.so.6.9
#5  0x00007fa6fb5974ad in dart::utils::DartLoader::parseSkeleton(dart::common::Uri const&, std::shared_ptr<dart::common::ResourceRetriever> const&) ()
   from /usr/local/lib/libdart-utils-urdf.so.6.9
#6  0x00007fa6fcc66e38 in robot_dart::Robot::_load_model (
    this=this@entry=0x55ca0102cfe0, filename="res/models/pexod.urdf", 
    packages=std::vector of length 0, capacity 0, 
    is_urdf_string=is_urdf_string@entry=false)
    at ../src/robot_dart/robot.cpp:1013
#7  0x00007fa6fcc6990b in robot_dart::Robot::Robot (this=0x55ca0102cfe0, 
    model_file="res/models/pexod.urdf", 
    packages=std::vector of length 0, capacity 0, robot_name=..., 
    is_urdf_string=<optimized out>, cast_shadows=<optimized out>, 
    damages=std::vector of length 0, capacity 0)
    at ../src/robot_dart/robot.cpp:151
#8  0x00007fa6fcc69c95 in robot_dart::Robot::Robot (this=0x55ca0102cfe0, 
    model_file="res/models/pexod.urdf", robot_name="robot", 
    is_urdf_string=<optimized out>, cast_shadows=<optimized out>, damages=...)
    at ../src/robot_dart/robot.cpp:157
#9  0x000055c9ff1d96c0 in __gnu_cxx::new_allocator<robot_dart::Robot>::construct<robot_dart::Robot, char const (&) [22]> (this=0x7ffeb510ad67, 
---Type <return> to continue, or q <return> to quit---
    __p=0x55ca0102cfe0) at /usr/include/c++/7/ext/new_allocator.h:136
#10 0x000055c9ff1d90d6 in std::allocator_traits<std::allocator<robot_dart::Robot> >::construct<robot_dart::Robot, char const (&) [22]> (__a=..., 
    __p=0x55ca0102cfe0) at /usr/include/c++/7/bits/alloc_traits.h:475
#11 0x000055c9ff1d8b5c in std::_Sp_counted_ptr_inplace<robot_dart::Robot, std::allocator<robot_dart::Robot>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<char const (&) [22]> (this=0x55ca0102cfd0, __a=...)
    at /usr/include/c++/7/bits/shared_ptr_base.h:526
#12 0x000055c9ff1d84a5 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<robot_dart::Robot, std::allocator<robot_dart::Robot>, char const (&) [22]> (this=0x7ffeb510af28, __a=...)
    at /usr/include/c++/7/bits/shared_ptr_base.h:637
#13 0x000055c9ff1d8014 in std::__shared_ptr<robot_dart::Robot, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<robot_dart::Robot>, char const (&) [22]> (this=0x7ffeb510af20, __tag=..., __a=...)
    at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#14 0x000055c9ff1d78d3 in std::shared_ptr<robot_dart::Robot>::shared_ptr<std::allocator<robot_dart::Robot>, char const (&) [22]> (this=0x7ffeb510af20, 
    __tag=..., __a=...) at /usr/include/c++/7/bits/shared_ptr.h:344
#15 0x000055c9ff1d6d48 in std::allocate_shared<robot_dart::Robot, std::allocator<robot_dart::Robot>, char const (&) [22]> (__a=...)
    at /usr/include/c++/7/bits/shared_ptr.h:691
#16 0x000055c9ff1d60cc in std::make_shared<robot_dart::Robot, char const (&) [2---Type <return> to continue, or q <return> to quit---
2]> () at /usr/include/c++/7/bits/shared_ptr.h:707
#17 0x000055c9ff1d515d in main () at src/examples/hexapod.cpp:11
(gdb) 
~~~

danachee, 2020-06-06:
Additional info (linking)

~~~
 ldd /tmp/hex
	linux-vdso.so.1 (0x00007fffa57eb000)
	libRobotDARTSimu.so => /usr/local/lib/libRobotDARTSimu.so (0x00007f56b6152000)
	libdart.so.6.9 => /usr/local/lib/libdart.so.6.9 (0x00007f56b5902000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f56b5579000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f56b5361000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f56b4f70000)
	libdart-utils.so.6.9 => /usr/local/lib/libdart-utils.so.6.9 (0x00007f56b4cba000)
	libdart-utils-urdf.so.6.9 => /usr/local/lib/libdart-utils-urdf.so.6.9 (0x00007f56b4a89000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f56b486a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f56b44cc000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f56b42c8000)
	libdart-external-odelcpsolver.so.6.9 => /usr/local/lib/libdart-external-odelcpsolver.so.6.9 (0x00007f56b40ba000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f56b3eb5000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f56b3c9b000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007f56b3993000)
	libfcl.so.0.5 => /usr/lib/x86_64-linux-gnu/libfcl.so.0.5 (0x00007f56b2eab000)
	libassimp.so.4 => /usr/lib/x86_64-linux-gnu/libassimp.so.4 (0x00007f56b24e0000)
	liboctomap.so.1.8 => /usr/lib/liboctomap.so.1.8 (0x00007f56b229c000)
	liboctomath.so.1.8 => /usr/lib/liboctomath.so.1.8 (0x00007f56b2096000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f56b65b8000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f56b1e82000)
	liburdfdom_model.so.1.0 => /usr/lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f56b1c60000)
	libicui18n.so.60 => /usr/lib/x86_64-linux-gnu/libicui18n.so.60 (0x00007f56b17bf000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007f56b1407000)
	libccd.so.2 => /usr/lib/x86_64-linux-gnu/libccd.so.2 (0x00007f56b11fc000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f56b0fdf000)
	libminizip.so.1 => /usr/lib/x86_64-linux-gnu/libminizip.so.1 (0x00007f56b0dd4000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f56b0bcc000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f56b09b7000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f56b07b2000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007f56aec09000)
~~~

danachee, 2020-06-06:
Ok, there's probably something missing somehow in a straight build.  But after crawling around, I found the files in cmake/example and those worked.  So I created a equivalent cmakefile for the src/examples and so far all is well.

Closing this.

closed 2020-06-06 by danachee

costashatz, 2020-06-06:
@danachee thank you for using our library. First of all, as we say in the README, the library is currently under heavy restructuring and many things (including the documentation/installation instructions) might not be up to date.

We will soon make a big v1.0 release with hopefully a proper documentation and surely many new features.

> Compiled:
> 
> ```
> g++ -g -I/usr/include/eigen3 -o /tmp/hex src/examples/hexapod.cpp -l RobotDARTSimu -ldart
> ```

If you want to compile it by hand, you would need to add many more flags and link to more libraries. We strongly suggest to use `waf` (our current favorite build system -- for an example usage for finding robot_dart can be found [here](https://github.com/costashatz/whc/blob/master/wscript) and [here](https://github.com/costashatz/whc/blob/master/waf_tools/robot_dart.py)). If you prefer CMake, we recently provided support for this, so when you do `[sudo] ./waf install` a CMake config file will be installed, so that one can use CMake to find RobotDART along with its dependencies.

I guess you already discovered this, but just in case..

danachee, 2020-06-06:
I realize its in flux, and I might just need to wait, but it seems the python side is "missing"??  I set it up to build the bindings, and it did install the object:

+ install /usr/local/lib/python3/dist-packages/RobotDART.cpython-38-x86_64-linux-gnu.so (from build/RobotDART.cpython-38-x86_64-linux-gnu.so)

But it didn't install anything else, and I can't find any python modules.  Are they missing, or do I need a different branch?

Thanks,

costashatz, 2020-06-06:
> But it didn't install anything else, and I can't find any python modules

You do not need anything else. Most probably you have to change your `PYTHONPATH` (I guess by default it is not looking to `/usr/local/`): `export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3/dist-packages/`. Alternatively, you can do `./waf configure --prefix=/usr --python` and then `sudo ./waf install`, this will install everything in `/usr` which should be okay for the `PYTHONPATH`. I prefer the first way (install in `/usr/local` and alter the `PYTHONPATH` as this keeps the system cleaner; you can also put the export in your bashrc/zshrc file so that you have it in every terminal by default).

danachee, 2020-06-06:
You are correct!  Thanks again! (it was a typeo in my PYTHONPATH set that was the issue)

About waf.  Is it something that would be used on a "outside" set of code (not stored in the robot_dart repo)?

So if one were to create their own simulation, do they need to copy the waf and waf_tools into their directory?  And if so, what would they need to tell it?  Feel free to tell me this will all be in the user guide for the new version and to be patient :-).

costashatz, 2020-06-06:
> You are correct! Thanks again! (it was a typeo in my PYTHONPATH set that was the issue)

Good!

> About waf. Is it something that would be used on a "outside" set of code (not stored in the robot_dart repo)?

You do not **need** to use waf. We have provide the installation of CMake config for this reason: for people to use robot_dart within their CMake projects easily. So if you are familiar with CMake and using it, feel free to use the library in that way.

> So if one were to create their own simulation, do they need to copy the waf and waf_tools into their directory? And if so, what would they need to tell it?

If you want to use waf for building your projects, there are many ways of doing this. I would suggest you to read the [waf book](https://waf.io/book/) where there is an extensive documentation of how to use waf in general.

> Feel free to tell me this will all be in the user guide for the new version and to be patient :-).

We will make a small standalone example of how to use waf similar to our use-cases to find robot_dart and compile your projects, so you can be a bit patient until the documentation is finished. If you want to have a look now on possible use-cases, have a look at the [whc library](https://github.com/costashatz/whc) that depends on robot_dart and uses waf to get an idea.