NOSALRO/robot_dart #86: bug with newest magnum and urdf_library branch

dalinel, 2020-07-15:
Hi,
I have updated the master branch of corrade / magnum / magnum-integration / magnum-plugins 
Taken the latest commit of the urdf_library branch

I have a core dump in the example , for the graphics.
Here is the backtrace :
```
#0  __memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:244
#1  0x00007ffff5d9524c in Corrade::Utility::Arguments::valueInternal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const () from /home/user/rf_ws/devel/lib/libCorradeUtility.so.2
#2  0x00007ffff6258d41 in Magnum::GL::Context::Context(Magnum::NoCreateT, Corrade::Utility::Arguments&, int, char const**, void (*)(Magnum::GL::Context&)) () from /home/user/rf_ws/devel/lib/libMagnumGL.so.2
#3  0x000000000046a2ea in Magnum::Platform::GLContext::GLContext(Magnum::NoCreateT, Corrade::Utility::Arguments&, int, char const**) ()
#4  0x000000000046a2a4 in Magnum::Platform::GLContext::GLContext(Magnum::NoCreateT, Corrade::Utility::Arguments&, int, char**) ()
#5  0x00000000004662cf in Magnum::Platform::GlfwApplication::GlfwApplication(Magnum::Platform::GlfwApplication::Arguments const&, Magnum::NoCreateT) ()
#6  0x0000000000441ea5 in robot_dart::gui::magnum::GlfwApplication::GlfwApplication (this=0x35df1d0, argc=0, argv=0x0, simu=0x7fffffffdc90, configuration=...) at ../src/robot_dart/gui/magnum/glfw_application.cpp:20
#7  0x0000000000420df4 in robot_dart::gui::magnum::make_application<robot_dart::gui::magnum::GlfwApplication> (configuration=..., simu=0x7fffffffdc90) at ../src/robot_dart/gui/magnum/base_application.hpp:204
#8  robot_dart::gui::magnum::BaseGraphics<robot_dart::gui::magnum::GlfwApplication>::BaseGraphics (configuration=..., simu=0x7fffffffdc90, this=<optimized out>) at ../src/robot_dart/gui/magnum/base_graphics.hpp:28
#9  robot_dart::gui::magnum::Graphics::Graphics (this=0x75e900, simu=0x7fffffffdc90, configuration=...) at ../src/robot_dart/gui/magnum/graphics.cpp:6
#10 0x0000000000420269 in __gnu_cxx::new_allocator<robot_dart::gui::magnum::Graphics>::construct<robot_dart::gui::magnum::Graphics, robot_dart::RobotDARTSimu*>(robot_dart::gui::magnum::Graphics*, robot_dart::RobotDARTSimu*&&) (__p=0x75e900, this=<optimized out>)
    at /usr/include/c++/5/ext/new_allocator.h:120
#11 std::allocator_traits<std::allocator<robot_dart::gui::magnum::Graphics> >::construct<robot_dart::gui::magnum::Graphics, robot_dart::RobotDARTSimu*>(std::allocator<robot_dart::gui::magnum::Graphics>&, robot_dart::gui::magnum::Graphics*, robot_dart::RobotDARTSimu*&&) (
    __p=<optimized out>, __a=...) at /usr/include/c++/5/bits/alloc_traits.h:530
#12 std::_Sp_counted_ptr_inplace<robot_dart::gui::magnum::Graphics, std::allocator<robot_dart::gui::magnum::Graphics>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<robot_dart::RobotDARTSimu*>(std::allocator<robot_dart::gui::magnum::Graphics>, robot_dart::RobotDARTSimu*&&) (__a=..., this=0x75e8f0) at /usr/include/c++/5/bits/shared_ptr_base.h:522
#13 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<robot_dart::gui::magnum::Graphics, std::allocator<robot_dart::gui::magnum::Graphics>, robot_dart::RobotDARTSimu*>(std::_Sp_make_shared_tag, robot_dart::gui::magnum::Graphics*, std::allocator<robot_dart::gui::magnum::Graphics> const&, robot_dart::RobotDARTSimu*&&) (__a=..., this=0x7fffffffdbd8) at /usr/include/c++/5/bits/shared_ptr_base.h:617
#14 std::__shared_ptr<robot_dart::gui::magnum::Graphics, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<robot_dart::gui::magnum::Graphics>, robot_dart::RobotDARTSimu*>(std::_Sp_make_shared_tag, std::allocator<robot_dart::gui::magnum::Graphics> const&, robot_dart::RobotDARTSimu*&&) (this=this@entry=0x7fffffffdbd0, __tag=..., __a=...) at /usr/include/c++/5/bits/shared_ptr_base.h:1097
#15 0x000000000041aab2 in std::shared_ptr<robot_dart::gui::magnum::Graphics>::shared_ptr<std::allocator<robot_dart::gui::magnum::Graphics>, robot_dart::RobotDARTSimu*>(std::_Sp_make_shared_tag, std::allocator<robot_dart::gui::magnum::Graphics> const&, robot_dart::RobotDARTSimu*&&) (__a=..., __tag=..., this=0x7fffffffdbd0) at /usr/include/c++/5/bits/shared_ptr.h:319
#16 std::allocate_shared<robot_dart::gui::magnum::Graphics, std::allocator<robot_dart::gui::magnum::Graphics>, robot_dart::RobotDARTSimu*>(std::allocator<robot_dart::gui::magnum::Graphics> const&, robot_dart::RobotDARTSimu*&&) (__a=...)
    at /usr/include/c++/5/bits/shared_ptr.h:620
#17 std::make_shared<robot_dart::gui::magnum::Graphics, robot_dart::RobotDARTSimu*>(robot_dart::RobotDARTSimu*&&) () at /usr/include/c++/5/bits/shared_ptr.h:636
#18 main () at ../src/examples/talos.cpp:33

```

costashatz, 2020-07-15:
> I have updated the master branch of corrade / magnum / magnum-integration / magnum-plugins
> Taken the latest commit of the urdf_library branch

Are you sure you have the latest commits and re-built, re-installed everything correctly? What OS are you on? Did you rebuild DART also?

I did this 2 days ago on two different machines (with different OSes) and had no issue.

costashatz, 2020-07-15:
> Here is the backtrace :

Btw, this seems like a `-march=native` flag issue. Are you sure you compiled everything with the same flags?

dalinel, 2020-07-15:
Ok, my mistake, I have installed everything from scratch in a new ubuntu16.04 docker and it works well

I did not manage to fix my previous install
Everything was compiled without avx

jbmouret, 2020-07-15:
should we close the bug?

dalinel, 2020-07-15:
yes it works for me

closed 2020-07-15 by costashatz