NOSALRO/robot_dart #89: Assimp version < 5.0 does not open Talos/iCub meshes

costashatz, 2020-08-12:
Interestingly, assimp version <5.0 cannot open our `glb` files for Talos and iCub. In theory assimp version >=4.1.0 should be able to open those. @jbmouret is it easy for you to investigate this?

jbmouret, 2020-08-25:
How common is version 4.10 vs 5.0?

I will have a look but it is likely that they improved the GLB support between 5.0 and 4.10. Since we are using the last Blender (and not our own code), it is unlikely that we can do anything (except maybe downgrade Blender).

If this is an issue, the easiest "fix" would be to go back to collada or stl (but without the smoothing of the normals), or provide both (bigger...).

jbmouret, 2020-08-25:
- Ubuntu 16.04 has 3.x
- Ubuntu 18.04 has 4.1
- Ubuntu 20.04 has 5.x

So, having 4.1 would allow us to support 18.04 more easily.

costashatz, 2020-08-25:
> except maybe downgrade Blender

Which Blender version did you use for this?

> If this is an issue, the easiest "fix" would be to go back to collada or stl (but without the smoothing of the normals), or provide both (bigger...).

I think we should first try to make the files with an older version of Blender. Or maybe Blender has some options for exporting glb files (e.g., a version of glb).

jbmouret, 2020-08-25:
I used Blender 2.83

costashatz, 2020-08-25:
> I used Blender 2.83

I have 2.79.. Can you send me the script to try myself?

jbmouret, 2020-08-25:
Here are the options for gltf:
[image]

No option except but we can note that:
- this is gltf 2.0
- there is a compression option ("using draco", which probably adds a dependency)

jbmouret, 2020-08-25:
Here is the script for Talos:

```python
import bpy
import os
import math

path = os.getcwd() + '/talos_description/meshes/'  # set this path
print("PATH:", path)

for root, dirs, files in os.walk(path):
    for f in files:
        print("f:", f, root)
        if f.endswith('-meshlab.dae') :
            dae_file = os.path.join(root, f)
            gltf_file = os.path.splitext(dae_file)[0].replace('-meshlab', '')
            #gltf_file = os.path.splitext(dae_file)[0]
            print(dae_file, "=>", gltf_file)

            bpy.ops.object.select_all(action='SELECT')
            bpy.ops.object.delete()
            bpy.ops.wm.collada_import(filepath=dae_file)
            #bpy.ops.import_mesh.stl(filepath=dae_file)

            # fix the smoothing
            for obj in bpy.data.objects:
                if obj.type == 'MESH':
                    obj.data.use_auto_smooth = 1
                    obj.data.auto_smooth_angle = math.radians(5) # 12 degrees

            bpy.ops.object.shade_smooth()

            # texture
            obj = bpy.context.object
            obj.data.materials.clear()                    
            for img in bpy.data.images:
                if not img.users:
                    bpy.data.images.remove(img)
                    
            bpy.ops.object.select_all(action='SELECT')
            bpy.ops.export_scene.gltf(filepath=gltf_file, ui_tab='MESHES')%   

```

For talos (but not for iCub), I also add to call meshlab first, using this script:

```xml

# meshlab
import subprocess
import os

# you need to add the meshlabserver to your PATH

path = os.getcwd() + '/talos_description/meshes/'  # set this path
print("PATH:", path)

def make_script():
    script = '''
    <!DOCTYPE FilterScript>
    <FilterScript>
    <filter name="Merge Close Vertices">
        <Param name="Threshold" value="1e-10" description="Merging distance" min="0" max="1" type="RichAbsPerc" />
    </filter>
    <filter name="Smooth Face Normals"/>
    <filter name="Transfer Color: Texture to Vertex"/>
    </FilterScript>
    '''

    f = open("/tmp/script.mxl", 'w')
    f.write(script)
    f.close()

def make_cmd(infile, outfile):
    return 'meshlabserver -i {} -o {} -m vc vn  -s /tmp/script.mxl'.format(infile, outfile).split(' ')

make_script()

for root, dirs, files in os.walk(path):
    for f in files:
        print("f:", f, root)
        if f.endswith('.dae') :
            dae_file = os.path.join(root, f)
            basename = os.path.splitext(dae_file)[0]
            dae_file_clean = basename + '-meshlab.dae'
            cmd = make_cmd(dae_file, dae_file_clean)
            subprocess.run(cmd)
```

for iCub:
```python
import bpy
import os
import math

path = os.getcwd() + '/icub_description/meshes/'  # set this path
print("PATH:", path)

for root, dirs, files in os.walk(path):
    for f in files:
        print("f:", f, root)
        if f.endswith('.dae') :
            dae_file = os.path.join(root, f)
            gltf_file = os.path.splitext(dae_file)[0]
            print(dae_file, "=>", gltf_file)

            bpy.ops.object.select_all(action='SELECT')
            bpy.ops.object.delete()
            bpy.ops.wm.collada_import(filepath=dae_file)

            # fix the smoothing
            for obj in bpy.data.objects:
                if obj.type == 'MESH':
                    obj.data.use_auto_smooth = 1
                    obj.data.auto_smooth_angle = math.radians(5) # 12 degrees

            bpy.ops.object.shade_smooth()
                    
            bpy.ops.object.select_all(action='SELECT')
            bpy.ops.export_scene.gltf(filepath=gltf_file, ui_tab='MESHES')% 
```

jbmouret, 2020-11-26:
More about this issue:

I installed assimp5 in my home directory (any install directory should work).

I compiled dart with:
`cmake -DCMAKE_INSTALL_PREFIX=$HOME -DASSIMP_INCLUDE_DIRS=$HOME/include -DASSIMP_LIBRARIES=$HOME/lib/libassimp.so ..
`
The key here is that we give the absolute path to assimp.

For robot_dart, `./waf configure --dart=$HOME`
=> we link with both assimp4/3 and assimp5 !

But we should not even try to link with assimp ourselves, since Dart is already linked to it (at least on linux).

=> I removed assimp from the waf_tools/dart.py 
https://github.com/resibots/robot_dart/blob/59d779a5400bd3c1596e32a133dc0ef249be2238/waf_tools/dart.py#L204

Now I can run the examples (at least the plain ones).

costashatz, 2021-09-27:
Did we solve this?

jbmouret, 2021-09-27:
No idea ;) I think assimp4 is still making troubles when it is installed, but maybe with Ubuntu 20.04 it is OK?

costashatz, 2021-09-27:
With Ubuntu 20.04 this is OK since assimp5 is by default installed there.

costashatz, 2021-09-27:
So I guess we ignore this and close this issue? Ubuntu 18.04 is already getting "old" and I am too bored to find a nicer solution other than compiling assimp from sources (which is easy to do).

jbmouret, 2021-09-28:
Yes, let's close it.

closed 2021-09-28 by costashatz