NOSALRO/robot_dart #104: Boost.process in camera does not compile with Sferes

Aneoshun, 2020-10-04:
Hi all, 

I am trying the latest version of Robot_dart in our dev pipeline and so far everything seems to work well. Thanks for all the new features!

The only thing that is causing issues at the moment is the boost.process part in camera.hpp. 
While robot_dart compiles without problem, my sferes experiment (a basic MAP-Elites hexapod one) does not compile. 
I am using ubuntu 20.04, boost 1.71, current sferes.QD branch, yesterday's Robot_Dart master branch. 

Have you observed the same issue on your side? Commenting out the #ifdef that defines ROBOT_DART_HAS_BOOST_PROCESS makes it compile, but that is not satisfactory. 

Here is the config messages and compilation errors:

```
Singularity example_dart_exp.sif:/git/sferes2> ./setup.sh -j1 -v
WARNING simplejson not found some function may not work
Command-line options for exp [exp/example_dart_exp] :  -> OK
Command-line options for exp [exp/example]          :  -> OK
Setting top to                           : /git/sferes2
Setting out to                           : /git/sferes2/build
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking boost includes                  : 1_71
Checking boost libs                      : ok
Checking Intel TBB includes (optional)   : /usr/include
Checking Intel TBB libs (optional)       : /usr/lib/x86_64-linux-gnu
Checking for MPI include (optional)      : Not found
Checking for Eigen                       : ok
Checking for ssrc kdtree (KD-tree)       : ok
Checking pthread                         : /usr/lib/x86_64-linux-gnu
Configuring for exp [example_dart_exp]
conf exp:
Checking for DART includes (including io/urdf) : 6.9.4 in /workspace/include
Checking for DART libs (including io/urdf)     : ['dart', 'dart-utils', 'dart-utils-urdf']
DART: Checking for Assimp                      : /usr/include
DART: Checking for Bullet Collision libs       : libs: ['LinearMath', 'BulletCollision', 'dart-collision-bullet'], bullet: /usr/include/bullet
DART: Checking for Ode Collision libs          : libs: ['ode', 'dart-collision-ode'], ode: /usr/include
Checking for Corrade includes                  : /workspace/include
Getting Corrade configuration                  : ['BUILD_DEPRECATED', 'TARGET_UNIX']
Checking for Corrade components                : ['CorradePluginManager', 'CorradeUtility', '/workspace/bin/corrade-rc']
Checking for Magnum includes                   : /workspace/include
Checking for Magnum lib                        : ['Magnum']
Getting Magnum configuration                   : ['BUILD_DEPRECATED', 'BUILD_MULTITHREADED', 'TARGET_GL']
Magnum: Checking for OpenGL includes           : /usr/include
Magnum: Checking for OpenGL lib                : ['GL']
Magnum: Checking for MagnumGL lib              : ['MagnumGL']
Checking for Magnum components                 : ['Shaders', 'Primitives', 'MagnumFont', 'TgaImporter', 'GlfwApplication', 'Trade', 'TextureTools', 'SceneGraph', 'WindowlessGlxApplication', 'MeshTools', 'Text']
Checking for AssimpImporter Magnum Plugin      : /workspace/include
AssimpImporter: Checking for Assimp            : /usr/include
Checking for AnyImageImporter Magnum Plugin    : /workspace/include
MagnumPlugins libs:                            : ['assimp']
Checking for Dart Magnum Integration           : /workspace/include
MagnumIntegration libs:                        : ['MagnumDartIntegration']
Checking for robot_dart includes               : ok: /workspace/include
Checking for robot_dart libs                   : libs: ['RobotDARTSimu', 'RobotDARTMagnum']
done
example_dart_exp -> ok

--- configuration ---
compiler(s):
 * CXX: gcc
boost version: 1_71
mpi: False
Compilation flags :
   CXXFLAGS : -D_REENTRANT -Wall -fPIC -ftemplate-depth-1024 -Wno-sign-compare -Wno-deprecated  -Wno-unused -DSFERES_ROOT="/git/sferes2"  -std=c++11 -DEIGEN3_ENABLED -DUSE_KDTREE
   LINKFLAGS:
--- license ---
Sferes2 is distributed under the CECILL license (GPL-compatible)
Please check the accompanying COPYING file or http://www.cecill.info/
'configure' finished successfully (0.093s)
WARNING simplejson not found some function may not work
Command-line options for exp [exp/example_dart_exp] :  -> OK
[... 102 lines cut ...]
                 from /usr/include/boost/asio/signal_set.hpp:19,
                 from ../sferes/run.hpp:50,
                 from ../exp/example_dart_exp/src/dart_exp.cpp:41:
/usr/include/boost/process/detail/posix/handles.hpp:78:12: error: request for member ‘st_mode’ in ‘stat_’, which is of non-class type ‘int’
   78 |     return S_ISCHR  (stat_.st_mode)  //This macro returns non-zero if the file is a character special file (a device like a terminal).
      |            ^~~~~~~
/usr/include/boost/process/detail/posix/handles.hpp:79:12: error: request for member ‘st_mode’ in ‘stat_’, which is of non-class type ‘int’
   79 |         || S_ISBLK  (stat_.st_mode) // This macro returns non-zero if the file is a block special file (a device like a disk).
      |            ^~~~~~~
/usr/include/boost/process/detail/posix/handles.hpp:80:12: error: request for member ‘st_mode’ in ‘stat_’, which is of non-class type ‘int’
   80 |         || S_ISREG  (stat_.st_mode) // This macro returns non-zero if the file is a regular file.
      |            ^~~~~~~
/usr/include/boost/process/detail/posix/handles.hpp:81:12: error: request for member ‘st_mode’ in ‘stat_’, which is of non-class type ‘int’
   81 |         || S_ISFIFO (stat_.st_mode) // This macro returns non-zero if the file is a FIFO special file, or a pipe. See section 15. Pipes and FIFOs.
      |            ^~~~~~~~
/usr/include/boost/process/detail/posix/handles.hpp:82:12: error: request for member ‘st_mode’ in ‘stat_’, which is of non-class type ‘int’
   82 |         || S_ISSOCK (stat_.st_mode) ;// This macro returns non-zero if the file is a socket. See section 16. Sockets.;
      |            ^~~~~~~~

Waf: Leaving directory `/git/sferes2/build'
Build failed
 -> task in 'example_graphic' failed with exit status 1:
        {task 140482117952000: cxx dart_exp.cpp -> dart_exp.cpp.2.o}
['/usr/bin/g++', '-D_REENTRANT', '-Wall', '-fPIC', '-ftemplate-depth-1024', '-Wno-sign-compare', '-Wno-deprecated', '-Wno-unused', '-DSFERES_ROOT="/git/sferes2"', '-std=c++11', '-DEIGEN3_ENABLED', '-DUSE_KDTREE', '-DVERSION="b\'f544c6fe37069935e5eedd2ba2f741c3a10c2a03\\n\'"', '-O3', '-mavx', '-msse', '-msse2', '-g', '-faligned-new', '-std=c++14', '-Iexp/example_dart_exp', '-I../exp/example_dart_exp', '-I.', '-I..', '-I/workspace/include', '-I/usr/include', '-I/workspace/include/MagnumExternal/OpenGL', '-I/usr/include/eigen3', '-Iexp/example_dart_exp/n', '-I../exp/example_dart_exp/n', '-I/', '-Iexp/example_dart_exp/i', '-I../exp/example_dart_exp/i', '-Iexp/example_dart_exp/r', '-I../exp/example_dart_exp/r', '-Iexp/example_dart_exp/u', '-I../exp/example_dart_exp/u', '-Iexp/example_dart_exp/l', '-I../exp/example_dart_exp/l', '-Iexp/example_dart_exp/d', '-I../exp/example_dart_exp/d', '-Iexp/example_dart_exp/c', '-I../exp/example_dart_exp/c', '-Iexp/example_dart_exp/e', '-I../exp/example_dart_exp/e', '-Iexp/example_dart_exp/s', '-I../exp/example_dart_exp/s', '-I/usr/include/bullet', '-DGRAPHIC', '-DMAGNUM_PLUGINS_DIR="/workspace/lib/magnum"', '-DMAGNUM_PLUGINS_FONT_DIR="/workspace/lib/magnum/fonts"', '-DMAGNUM_PLUGINS_FONTCONVERTER_DIR="/workspace/lib/magnum/fontconverters"', '-DMAGNUM_PLUGINS_IMAGECONVERTER_DIR="/workspace/lib/magnum/imageconverters"', '-DMAGNUM_PLUGINS_IMPORTER_DIR="/workspace/lib/magnum/importers"', '-DMAGNUM_PLUGINS_AUDIOIMPORTER_DIR="/workspace/lib/magnum/audioimporters"', '-DBUILD_DEPRECATED', '-DBUILD_MULTITHREADED', '-DTARGET_GL', '-DMAGNUM_PLUGINS_DIR="/workspace/lib/magnum"', '-DMAGNUM_PLUGINS_FONT_DIR="/workspace/lib/magnum/fonts"', '-DMAGNUM_PLUGINS_FONTCONVERTER_DIR="/workspace/lib/magnum/fontconverters"', '-DMAGNUM_PLUGINS_IMAGECONVERTER_DIR="/workspace/lib/magnum/imageconverters"', '-DMAGNUM_PLUGINS_IMPORTER_DIR="/workspace/lib/magnum/importers"', '-DMAGNUM_PLUGINS_AUDIOIMPORTER_DIR="/workspace/lib/magnum/audioimporters"', '-DBUILD_DEPRECATED', '-DBUILD_MULTITHREADED', '-DTARGET_GL', '-DMAGNUM_PLUGINS_DIR="/workspace/lib/magnum"', '-DMAGNUM_PLUGINS_FONT_DIR="/workspace/lib/magnum/fonts"', '-DMAGNUM_PLUGINS_FONTCONVERTER_DIR="/workspace/lib/magnum/fontconverters"', '-DMAGNUM_PLUGINS_IMAGECONVERTER_DIR="/workspace/lib/magnum/imageconverters"', '-DMAGNUM_PLUGINS_IMPORTER_DIR="/workspace/lib/magnum/importers"', '-DMAGNUM_PLUGINS_AUDIOIMPORTER_DIR="/workspace/lib/magnum/audioimporters"', '-DBUILD_DEPRECATED', '-DBUILD_MULTITHREADED', '-DTARGET_GL', '-DUSE_TBB', '../exp/example_dart_exp/src/dart_exp.cpp', '-c', '-o/git/sferes2/build/exp/example_dart_exp/src/dart_exp.cpp.2.o']
```

costashatz, 2020-10-06:
Hello @Aneoshun! Thanks!

Can you give us the code of your experiment? I think that this might have to do with the order of the include headers.

jbmouret, 2020-10-07:
Hi

I think the issue is from Sferes: we have a namespace named `stat` and it conflicts with the stat function (system call). I have run in this issue a few times already and I played with includes (e.g. including `#include <sys/stat.h>` or things like #undef stat, etc.). The issue is here:
https://github.com/sferes2/sferes2/issues/103 

It is clear to me that boost.process is likely to include a call to stat at some point, which would explain the issue.

I have a nsga-2 + robot_dart that works. 

About the boost.process: it is currently optional. If we do not have it, we can still record videos but it is using classic unix pipes (slightly less portable and less elegant).

=> I think we should rename the namespace in sferes...

costashatz, 2020-10-10:
@Aneoshun can you verify that the issue is not coming from robot_dart to close this issue?

Aneoshun, 2020-12-15:
@costashatz Sorry for the delay. I agree with @jbmouret, the issue is likely to come from Sferes. Robot_dart works well on its own. I will close the issue.

closed 2020-12-15 by Aneoshun