NOSALRO/robot_dart #133: Unable to set damages to robot.

IIIIIIIIll, 2021-05-21:
Hi, I was trying to use robot_dart's new API to replicate the experiment in the intelligent trial and error paper. But I was unable to set damages to the hexapod robot either using the existing code from the experiment or using the examples shown [here](https://github.com/resibots/robot_dart/blob/master/docs/guide.md). 

**To Reproduce**
- Based on old intelligent trial and error algorithm code
If this method is used then the robot will freeze in experiment after initialization(visualization provided as an attachment).

https://user-images.githubusercontent.com/21286615/119079671-2316f800-ba2b-11eb-8d69-5e7e3af86cca.mp4

I have tried to set all joints to passive and the robot would collapse as expected. Also if no damage is present the robot would move normally as expected. I can confirm the dmg.data contains the correct name and the joint is handled correctly by the simulations such that the correct joint name could be retrieved by replacing `setActuatorType(dart::dynamics::Joint::PASSIVE)` to `getName()` 
```
  global::global_robot = std::make_shared<robot_dart::Robot>("exp/singularity_hexapod_adaption/resources/hexapod_v2.urdf");
  global::global_robot->set_position_enforced(true);
  global::global_robot->set_actuator_types("servo");
  set_damages(damages);
 
 //In set_damages() this specific line is used
  global::global_robot->skeleton()->getJoint(dmg.data)->setActuatorType(dart::dynamics::Joint::PASSIVE);
```

- Based on examples given in the guide.
If the new API constructor is used then no damage is set to the robot even if damages are specified

```
  global::global_robot = std::make_shared<robot_dart::Robot>("exp/singularity_hexapod_adaption/resources/hexapod_v2.urdf", "robot", false, true, damages);
  //Here damages is of type std::vector<robot_dart::RobotDamage>
  global::global_robot->set_position_enforced(true);
  global::global_robot->set_actuator_types("servo");
```

**Things I have tried**
- Change joint type from "PASSIVE" to "LOCKED"
The robot also freezes right after initialization, not responding to simulation. No action of robot is observed, no joint movement observed.
- Set all but one joint to "PASSIVE"
The robot would behave differently from the situation that all joints are set to PASSIVE. The leg which the joint set to SERVO belongs to would stretch outward instead of collapsing like other legs. 

I could provide more details if that's needed, but so far that is all the experiments and tests I have done. Thank you!

costashatz, 2021-05-21:
@IIIIIIIIll thanks for using our library.

First of all, we are planning to remove the damage interface. We are not actively using it and we give all the tools to the user to insert damages without the need of a specific structure/definition.

> * Based on examples given in the guide.
>   If the new API constructor is used then no damage is set to the robot even if damages are specified

I just tried this and this is NOT working as expected. This is because `set_actuator_types` overrides the locked joint. You can try something like the following:

```cpp
auto robot = std::make_shared<robot_dart::Robot>("pexod.urdf");

robot->set_position_enforced(true);
robot->set_actuator_types("servo");

// the next 2 lines have to come after robot->set_actuator_types("servo");
robot->set_positions(robot_dart::make_vector({2.}), {"link_name"});
robot->set_actuator_type("locked", "link_name");
```

I just tried this and works as expected. However you need to be careful when controlling the robot. The `RobotControl` class will *remove* the locked joints from the available degrees of freedom (and thus return fewer degrees of freedom). You can either pass your (desired) controllable dofs the `RobotControl` class or write your own controller without using `RobotControl`. See [this example](https://github.com/resibots/robot_dart/blob/master/src/examples/control_loop.cpp) for how to do a control loop without the need of `RobotControl`.

jbmouret, 2021-05-21:
Just to be sure: the current example code does not attempt to move the legs (it was previously, but we removed the controller as it was unrelated code). Are you sure you are moving the legs with a  controller?

IIIIIIIIll, 2021-05-21:
Yes I am, later in the code I have this piece of code. The controller is called using a cpp template to the `policy_control.hpp` file just like the one being used in the it&e source code. 
```
g_robot->add_controller(std::make_shared<robot_dart::control::HexaControl>(ctrl_dt, ctrl));
std::static_pointer_cast<robot_dart::control::HexaControl>(g_robot->controllers()[0])->set_h_params(Eigen::VectorXd::Ones(1) * ctrl_dt);
std::static_pointer_cast<robot_dart::control::HexaControl>(g_robot->controllers()[0])->activate(false);
robot_dart::RobotDARTSimu simu(ctrl_dt);
......
std::static_pointer_cast<robot_dart::control::HexaControl>(g_robot->controllers()[0])->activate(true);
```
And the robot moves without any issue if no damage is applied.

IIIIIIIIll, 2021-05-21:
I tried to play around the `policy_control.hpp` file a bit and found when damage is applied the controller won't use the query function from the provided policy like it normally would. But I did not get any complains from the assertion in `Eigen::VectorXd calculate(double t)`. It seems the function is not called.

costashatz, 2021-05-21:
> I tried to play around the `policy_control.hpp` file a bit and found when damage is applied to the controller, it won't use the query function from the provided policy like it normally would. But I did not get any complains from the assertion in `Eigen::VectorXd calculate(double t)` which is the only place where it was used. It seems the function is not called.

Yes this is normal. When you lock one joint, you remove one DOF. This makes the check in `HexaControl` to fail and thus the controller is never enabled. You just need to check in the controller that you include the also the locked joints (see the lines [here](https://github.com/resibots/robot_dart/blob/master/src/robot_dart/control/robot_control.cpp#L32) and [here](https://github.com/resibots/robot_dart/blob/master/src/robot_dart/control/robot_control.cpp#L37)).

In your video, the robot does not freeze. The controller is never enabled and thus the robot stays in position (achieves zero velocity with the servo actuators).

Hope that I helped!

IIIIIIIIll, 2021-05-21:
Thank you for your help! I am able to simulate damaged robots now. I will close the issue 😄

closed 2021-05-21 by IIIIIIIIll

costashatz, 2021-05-21:
I am glad! Let us know if you encounter any other issue.