NOSALRO/robot_dart #135: Adapt to DART 7

costashatz, 2021-07-18:
DART master/main branch is on v7. There are many things that have changed compared to DART 6. We need to adapt robot_dart to this..

jbmouret, 2021-07-19:
Is there a list of improvements somewhere? what is better in dart7?

costashatz, 2021-07-19:
> Is there a list of improvements somewhere?

No. I asked and they said that they are planning to do one. One can view the change log once it is released for high-level changes.

> what is better in dart7?

It's not about being better. DART 6.11.0 (or 6.12.0) will be the last release with DART 6 API and then abandoned. DART 7 is the one that is going to be developed/maintained. The main changes are mostly restructuring and removal of parts, and thus it is changing API quite a bit (also CMake).

We do not have to be super fast for this (they just released 6.11, so it'll take a while to reach v7), but the sooner the better as otherwise we are going to lose track.

As of now, robot_dart and the master branch of DART do not work together.

costashatz, 2021-09-24:
FYI: DART master has switched back to v6+. DART 7 is at dart7-* branches. Closing this for now..

closed 2021-09-24 by costashatz