NOSALRO/robot_dart #187: `libdart.so` not found

im-Kitsch, 2022-07-18:
Hi,

I am just beginner of robo_dart. I followed the installation guid in ReadMe, but I failed at step `./waf  examples`, the output is as follows, it seems `libdart.so` is not generated, is there any reason or any method that I can locate the problem? I guess maybe the building process has some problems, but if libdart.so is not generated, there should be some errors while building, but I don't get the reason.   

Any hint would be very appreciated! Thanks a lot!

```
 ./waf examples
Waf: Entering directory `/pkgs/robot_dart/build'
=== Building RobotDART ===
Bulding examples...
[ 43/119] Linking build/ur3e_plain
[ 44/119] Linking build/ur3e
[ 47/119] Linking build/icub_plain
[ 48/119] Linking build/icub
[ 50/119] Linking build/dof_maps_plain
[ 52/119] Linking build/dof_maps
[ 54/119] Linking build/pendulum_plain
[ 56/119] Linking build/pendulum
[ 58/119] Linking build/transparent
[ 60/119] Linking build/talos_plain
[ 62/119] Linking build/talos
[ 64/119] Linking build/iiwa_plain
[ 66/119] Linking build/iiwa
[ 68/119] Linking build/scheduler_plain
[ 70/119] Linking build/magnum_contexts
[ 72/119] Linking build/iiwa_torque_plain
/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

/usr/bin/ld: warning: libdart-external-odelcpsolver.so.6.12, needed by /pkgs/install/lib/libdart.so, not found (try using -rpath or -rpath-link)
[... 63 lines cut ...]
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dFactorLDLT(double*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dRandInt(int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::_dSolveLDLT(double const*, double const*, double*, int, int)'
/pkgs/install/lib/libdart.so: undefined reference to `dart::external::ode::dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*, bool)'
collect2: error: ld returned 1 exit status

Waf: Leaving directory `/pkgs/robot_dart/build'
Build failed
 -> task in 'scheduler_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'ur3e_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'pendulum_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'dof_maps_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'icub_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'talos_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'iiwa_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'iiwa_torque_plain' failed with exit status 1 (run with -v to display more information)
 -> task in 'pendulum' failed with exit status 1 (run with -v to display more information)
 -> task in 'transparent' failed with exit status 1 (run with -v to display more information)
 -> task in 'dof_maps' failed with exit status 1 (run with -v to display more information)
 -> task in 'magnum_contexts' failed with exit status 1 (run with -v to display more information)
 -> task in 'iiwa' failed with exit status 1 (run with -v to display more information)
 -> task in 'ur3e' failed with exit status 1 (run with -v to display more information)
 -> task in 'icub' failed with exit status 1 (run with -v to display more information)
 -> task in 'talos' failed with exit status 1 (run with -v to display more information)
```

costashatz, 2022-07-18:
Thanks for using our library :)

- `sudo apt install mlocate`
- `sudo updatedb`
- `locate libdart-external-odelcpsolver.so`

What does it output?

Also what is the output of the following:

Inside the `robot_dart` folder
- `./waf distclean`
- `./waf configure`
- `./waf`

?

M-Amin-Ali, 2022-07-19:
Hi,
I use the branch 'new_docs' with the quick building and have the same issue!
Did you resolve this?
Let me know, pls!
Thanks

costashatz, 2022-07-19:
Hi @M-Amin-Ali 

First, the `new_docs` branch is not finished yet and most probably not fully compiling. Please use the master branch as of now.

Secondly, did you try running the following? (as suggested to @im-Kitsch )

- `sudo apt install mlocate`
- `sudo updatedb`
- `locate libdart-external-odelcpsolver.so`

What does it output?

Also what is the output of the following:

Inside the robot_dart folder

- `./waf distclean`
- `./waf configure`
- `./waf`
-

im-Kitsch, 2022-07-19:
HI,

thanks for the prompt reply and sorry for the late respone. 

After I run the command: 
```
sudo apt install mlocate
sudo updatedb
locate libdart-external-odelcpsolver.so
```
it shows as follows for `locate libdart-external-odelcpsolver.so`
```
root@76eaa694866f:/pkgs/robot_dart# locate libdart-external-odelcpsolver.so
/pkgs/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so
/pkgs/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so.6.12
/pkgs/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so.6.12.1
/pkgs/install/lib/libdart-external-odelcpsolver.so
/pkgs/install/lib/libdart-external-odelcpsolver.so.6.12
/pkgs/install/lib/libdart-external-odelcpsolver.so.6.12.1
/usr/lib/x86_64-linux-gnu/libdart-external-odelcpsolver.so
/usr/lib/x86_64-linux-gnu/libdart-external-odelcpsolver.so.6
/usr/lib/x86_64-linux-gnu/libdart-external-odelcpsolver.so.6.9.2
```

Then `./waf distclean` it shows `'distclean' finished successfully (0.015s)`.

Instead of `./waf configure`, I run configuration with some specification of path, command and output are as follows:

```
root@76eaa694866f:/pkgs/robot_dart# ./waf configure --dart=/pkgs/install --corrade_install_dir=/pkgs/install --magnum_install_dir=/pkgs/install --magnum_integration_install_dir=/pkgs/install --magnum_plugins_install_dir=/pkgs/install --prefix=/pkgs/install
Setting top to                           : /pkgs/robot_dart
Setting out to                           : /pkgs/robot_dart/build
=== Configuring RobotDART ===
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking boost includes                  : 1_71
Checking boost libs                      : ok
Checking for library ['pthread']         : yes
Checking for Eigen                       : /usr/include/eigen3
Checking for Eigen version               : 3.3.7
Checking for DART includes (including io/urdf) : 6.12.1 in /pkgs/install/include
Checking for DART libs (including io/urdf)     : ['dart', 'dart-utils', 'dart-utils-urdf']
DART: Checking for Bullet Collision libs       : libs: ['LinearMath', 'BulletCollision', 'dart-collision-bullet'], bullet: /usr/include/bullet
DART: Checking for Ode Collision libs          : libs: ['ode', 'dart-collision-ode'], ode: /usr/include
DART: Checking for Octomap libs                : libs: ['octomap'], octomap: /usr/include
Checking for Corrade includes                  : /pkgs/install/include
Getting Corrade configuration                  : ['BUILD_DEPRECATED', 'TARGET_UNIX']
Checking for Corrade components                : ['CorradePluginManager', 'CorradeUtility', '/pkgs/install/bin/corrade-rc']
Checking for Magnum includes                   : /pkgs/install/include
Checking for Magnum lib                        : ['Magnum']
Getting Magnum configuration                   : ['BUILD_DEPRECATED', 'BUILD_MULTITHREADED', 'TARGET_GL']
Magnum: Checking for OpenGL includes           : /usr/include
Magnum: Checking for OpenGL lib                : ['GL']
Magnum: Checking for MagnumGL lib              : ['MagnumGL']
Checking for Magnum components                 : ['TgaImporter', 'MagnumFont', 'GlfwApplication', 'Text', 'MeshTools', 'Trade', 'Primitives', 'Shaders', 'WindowlessGlxApplication', 'TextureTools', 'SceneGraph']
Checking for StbTrueTypeFont Magnum Plugin     : /pkgs/install/include
Checking for AnyImageImporter Magnum Plugin    : /pkgs/install/include
Checking for AssimpImporter Magnum Plugin      : /pkgs/install/include
AssimpImporter: Checking for Assimp            : /usr/include
MagnumPlugins libs:                            : ['assimp']
Checking for Eigen Magnum Integration          : /pkgs/install/include
Checking for Dart Magnum Integration           : /pkgs/install/include
MagnumIntegration libs:                        : ['MagnumDartIntegration']
AVX compilation of dart                        : no
AVX compilation of dart-utils                  : no
AVX compilation of dart-utils-urdf             : no
-march=native (AVX support)                    : no (optional)
['-g', '-faligned-new', '-Wmissing-declarations', '-Werror=return-type', '-pedantic', '-O3', '-Winit-self', '-Wall', '-Wextra', '-fPIC', '-std=c++14', '-Wold-style-cast']
=== RobotDART ready to build ===

[... 244 lines cut ...]
                 from ../src/robot_dart/gui/magnum/base_application.hpp:10,
                 from ../src/robot_dart/gui/magnum/sensor/camera.hpp:4,
                 from ../src/robot_dart/gui/magnum/sensor/camera.cpp:1:
../../install/include/Magnum/Shaders/VertexColor.h:44:13: warning: use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead
   44 | CORRADE_DEPRECATED_FILE("use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead")
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from ../src/robot_dart/gui/magnum/gs/camera.hpp:18,
                 from ../src/robot_dart/gui/magnum/gs/camera.cpp:3:
../../install/include/Magnum/Shaders/VertexColor.h:44:13: warning: use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead
   44 | CORRADE_DEPRECATED_FILE("use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead")
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[... cut ...]
../../install/include/Magnum/Shaders/VertexColor.h:44:13: warning: use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead
   44 | CORRADE_DEPRECATED_FILE("use Magnum/Shaders/VertexColorGL.h, the VertexColorGL class and related typedefs instead")
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[41/41] Linking build/libRobotDARTMagnum.a
Waf: Leaving directory `/pkgs/robot_dart/build'
'build' finished successfully (20.715s)
```

im-Kitsch, 2022-07-19:
Hi, @costashatz 

After I run the command you have suggested,  I re-run `./waf examples`, there is no error that is related to `libdart.so` ,  but there is still some errors that shows `undefined reference`. Is there any hint that is related to this error?

Thanks for the help
Best regards.

costashatz, 2022-07-19:
> /pkgs/install/lib/libdart-external-odelcpsolver.so
> /pkgs/install/lib/libdart-external-odelcpsolver.so.6.12
> /pkgs/install/lib/libdart-external-odelcpsolver.so.6.12.1 [...]

You have multiple versions of dart installed. This most probably means that waf finds one (e.g. 6.12.1) and then you haven't setup your `LD_LIBRARY_PATH` correctly and the linker attempts to link to the other (e.g., 6.9.2). Please remove them all and install just one from scratch.

Also if you install dart into a non-global path (e.g. at `/pkgs/install/`), you need to update your `LD_LIBRARY_PATH` to include the installed paths (e.g. `export LD_LIBRARY_PATH=/pkgs/install/lib:$LD_LIBRARY_PATH` in your `~/.bashrc` or `~/.zshrc` before configuring/building robot_dart).

M-Amin-Ali, 2022-07-20:
Hi @costashatz,
in my case:
/home/amin/install/robot_dart/temp_robot_dart/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so
/home/amin/install/robot_dart/temp_robot_dart/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so.6.12
/home/amin/install/robot_dart/temp_robot_dart/dart/build/dart/external/odelcpsolver/libdart-external-odelcpsolver.so.6.12.1
/opt/dart/lib/libdart-external-odelcpsolver.so
/opt/dart/lib/libdart-external-odelcpsolver.so.6.12
/opt/dart/lib/libdart-external-odelcpsolver.so.6.12.1
So I've not multiple versions of dart, but still to check that with master branch.

im-Kitsch, 2022-07-20:
Hi, @costashatz 

thanks for the hint, I just realized it's strange that I installed it in /usr/lib/x86_64-linux-gnu, I guess maybe I made any errors by the installation. But anyway, I restarted a new docker and add installation path to LD_LIBRARY_PATH and it works. 

I have not tested to remove redundant library libdart.so in /usr/lib. But I guess maybe just add LD_LIBRARY_PATH is enough. 

Thanks a lot for the help.

costashatz, 2022-07-20:
> **After new installation (dart v6.12.1 / python3.7 / gcc9.4):**
> /usr/bin/ld: warning: liboctomap.so.1.9, needed by /opt/dart/lib/libdart.so, may conflict with liboctomap.so.1.8

Have you set the `LD_LIBRARY_PATH` correctly so that `libdart.so` links to the correct version of octomap?

costashatz, 2022-07-20:
> I have not tested to remove redundant library libdart.so in /usr/lib. But I guess maybe just add LD_LIBRARY_PATH is enough.

I would delete the duplicate installation, but if set correctly `LD_LIBRARY_PATH` should suffice..

costashatz, 2022-07-20:
**Closing this as it seems resolved**

closed 2022-07-20 by costashatz

im-Kitsch, 2022-07-20:
> > I have not tested to remove redundant library libdart.so in /usr/lib. But I guess maybe just add LD_LIBRARY_PATH is enough.
> 
> I would delete the duplicate installation, but if set correctly `LD_LIBRARY_PATH` should suffice..

Hi, @costashatz 

it seems I made a wrong reply. I re-opened a docker and tested re-build it. Now I find that I have to remove the duplicated installation. I think last time that I succed without remove duplicated installtion relates to something like Cmake-Cache.

Now I know that the duplicated installation of `libdart*` is automatically installed from `apt-get`, since some ros-packages are dependent on that. So removing the "duplicated libdart" may be not a good solution since it should be re-installed again.

So my question is, how could I specify the verion or path of libDart?  multiple installation and specify a specific version should be normal in Linux, although it looks messy and annoying.  Do you have any hints for that? I think it should be a very normal problem in deployment and development. So a more safe solution would be to specify version/path of dependencies. 

I am sorry i am just beginner for Make and Cmake. It would be very very appreciated if you can give any hint. 

Best regards, thanks in advance.

costashatz, 2022-07-20:
> Now I know that the duplicated installation of `libdart*` is automatically installed from `apt-get`, since some ros-packages are dependent on that. So removing the "duplicated libdart" may be not a good solution since it should be re-installed again.

You can delete the one you built from source and use the one provided by the packages. robot_dart does not care. This would make it easier for you..

costashatz, 2022-07-20:
> It seems robot_dart explicitly required the built-from-source installation, i.e. v6.12.1, yes?

Nope... The documentation says it is advised to use robot_dart with v6.12.1, but robot_dart should work with any dart >= 6 and < 7.

costashatz, 2022-07-20:
Or at least, we wrote the docs for ourselves :stuck_out_tongue:

im-Kitsch, 2022-07-20:
> Or at least, we wrote the docs for ourselves 😛

Oh, cool, I will test it as soon as possible. 

Thanks a lot for the kind and patience! 👍

costashatz, 2022-07-20:
We will shortly (less than 2-3 months) release a new version of the docs along with an official v1.0 release of robot_dart. Then everything should be easier to handle :)