NOSALRO/robot_dart #202: Installation fails on Ubuntu 24.04 ARM64

c-joly, 2024-06-18:
Hello @costashatz, here is an issue linked to the Joss review https://github.com/openjournals/joss-reviews/issues/6771

I tried the installation script on an Ubuntu Virtual Machine 24.04 with arm64 architecture, and it looks to have a problem in the `apt` dependencies. I attached the terminal log here: https://gist.github.com/c-joly/3d68d729b40c51aa6b11231771941435 (note: I changed the script to make it stop at the first error)

Note: the install script **passed flawlessly on Ubuntu 22.04 arm64**

It looks like the `apt` dependencies are not satisfied on the last version of Ubuntu (at least for arm64 architecture, I can't test for amd64 with my Mac M2). Maybe you should consider to limit the versions of Ubuntu that are fine, >= 20.04 may be too optimistic.

costashatz, 2024-06-18:
Hey @c-joly! Thanks for testing.

Interesting that the DART developers have pre-built binaries for 22.04 and not for 24.04 (at least for arm64). I will test on 24.04 on an amd64 arch and see what I get. Then I'll update the script or the docs accordingly.

Good to know that Ubuntu 22.04 worked easily (even on arm64)!

Thanks again!

boisgera, 2024-06-20:
Hi @costashatz!

AFAICT, I the same issue also exists for x86, see below ; that makes sense, the 24.04 release is still rather new. So the title of this issue could be updated.

Small nitpicking : could you modify the `install_ubuntu.sh` script so that it stop as soon as an error is met? 
(see for example [Unofficial bash strict mode](https://gist.github.com/robin-a-meade/58d60124b88b60816e8349d1e3938615)).

Cheers,

Sébastien

```
boisgera@box:~/robot_dart$ ./scripts/install_ubuntu.sh
[sudo] password for boisgera:
PPA publishes dbgsym, you may need to include 'main/debug' component
Repository: 'Types: deb
URIs: https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu/
Suites: noble
Components: main
'
More info: https://launchpad.net/~dartsim/+archive/ubuntu/ppa
Adding repository.
Hit:1 http://archive.ubuntu.com/ubuntu noble InRelease
Get:2 http://archive.ubuntu.com/ubuntu noble-updates InRelease [126 kB]        
Hit:3 http://security.ubuntu.com/ubuntu noble-security InRelease              
Ign:4 https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble InRelease      
Err:5 https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble Release
  404  Not Found [IP: 185.125.190.80 443]
Hit:6 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Ign:1 https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble InRelease
Err:2 https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble Release        
  404  Not Found [IP: 185.125.190.80 443]
Hit:3 http://archive.ubuntu.com/ubuntu noble InRelease                        
Hit:4 http://security.ubuntu.com/ubuntu noble-security InRelease              
Hit:5 http://archive.ubuntu.com/ubuntu noble-updates InRelease
Hit:6 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/dartsim/ppa/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

costashatz, 2024-06-20:
@boisgera good to know! I'll inform the DART developers and update the docs until they are up to date. I know they are being very meticulous for the ROS supported Ubuntu versions. So I guess ROS is not on 24.04 yet..

costashatz, 2024-08-26:
@c-joly @boisgera can you both try again? It seems that DART is now available in 24.04 (see [here](https://www.ubuntuupdates.org/package/core/noble/universe/base/libdart-all-dev)).

boisgera, 2024-08-26:
On buntu 22.04, I do have an unexpected end of file error message.

The same error will appear (faster) if I only check the syntax of the install script, without running it

    ~/tmp/robot_dart$ bash -n ./scripts/install_ubuntu.sh 
    ./scripts/install_ubuntu.sh: line 106: syntax error: unexpected end of file

boisgera, 2024-08-26:
There is a missing `fi` in the script; when it's added, it passes the syntax check. I am trying the install again right now.

costashatz, 2024-08-26:
> There is a missing `fi` in the script; when it's added, it passes the syntax check. I am trying the install again right now.

I just pushed in the `master` branch, the corrected script (putting the `fi` in the correct place as it could be ambiguous).

Thank you for your effort :)

boisgera, 2024-08-26:
I have this error quite late in the build:
    
    Cloning into 'magnum-bindings'...
    remote: Enumerating objects: 5937, done.
    remote: Counting objects: 100% (1328/1328), done.
    remote: Compressing objects: 100% (466/466), done.
    remote: Total 5937 (delta 970), reused 1186 (delta 862), pack-reused 4609 (from 1)
    Receiving objects: 100% (5937/5937), 1.64 MiB | 1.46 MiB/s, done.
    Resolving deltas: 100% (4459/4459), done.
    -- The CXX compiler identification is GNU 13.2.0
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Check for working CXX compiler: /usr/bin/c++ - skipped
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
      Could NOT find Corrade (missing: CORRADE_INCLUDE_DIR
      _CORRADE_CONFIGURE_FILE)
    Call Stack (most recent call first):
      /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
      modules/FindCorrade.cmake:312 (find_package_handle_standard_args)
      modules/FindMagnum.cmake:262 (find_package)
      CMakeLists.txt:59 (find_package)
    
    
    -- Configuring incomplete, errors occurred!

costashatz, 2024-08-26:
> I have this error quite late in the build:

OK! Let me have a deeper look and come back to it. At least it seems that the Ubuntu 24.04 issue is gone. I'll fix the rest! Thanks! I'll ping you once I have this ready.

costashatz, 2024-08-28:
@boisgera @c-joly the `master` branch now contains an `install_ubuntu.sh` script that should work for >=20.04. I just tried in clean docker images.

Can you please try again and confirm this?

Thanks again!

costashatz, 2024-08-29:
> However, I noticed some problems with some examples. I experienced errors due to setScale, for examples with all examples with Talos and Tiago. @boisgera , @costashatz : can you check on x64 architecture?

I checked and it appears that this is coming from DART (it also happens in x64). They have tightened their checks for mesh scaling. Imho, they should allow negative ones (this just mirrors the mesh; very common practice in graphics community to put negative scale). I will contact them and see how we can solve this. Not very important, but annoying to not be able to use the basic examples. Thanks for catching this.

costashatz, 2024-09-03:
I think this is ready to be closed! @c-joly let me know if you find something else that needs fixing. Thanks again!

closed 2024-09-03 by costashatz