NOSALRO/robot_dart #203: Documentation for building C++ examples

c-joly, 2024-06-20:
Hi @costashatz, here is a new issue about the JOSS review https://github.com/openjournals/joss-reviews/issues/6771 
These comments are based on the Ubuntu 22.04-arm64 installation with the one line installation script

1- First, I successfully ran the Hello World example in Python, and I was happy to see my first robot in the graphical interface 😃

2- However, building the C++ example was a little bit more difficult. I did not find any building instructions on the FAQ page, so I started to manually compile the cpp example, which was a bad idea! Then, I figured out there is a CMake example in the source of the GitHub, but I did not see any reference to it in the doc (or maybe I scanned too fast the doc page and I missed it?)

3 - Finally, the example was not exactly working out of the box since `cmake` was not able to find `robot_dart`, `magnum`and `corrade`. I had to manually set the `CMAKE_PREFIX_PATH` environment variable:

  - `export CMAKE_PREFIX_PATH=/opt/magnum:/opt/robot_dart`
  - then, the "classical" building procedure with `cmake` works fine:
    - `mkdir build`
    - `cd build`
    - `cmake ..`
    - `make` (eventually with `-j4`, `-j8` depending on the computer capabilities...)
  - Then, I successfully adapted the `CMakeList.txt` to build an other example in headless and graphical mode. 
  - **Can you put instructions in the installation doc to add the `export CMAKE_PREFIX_PATH` command in the `.bashrc` / `.profile` / `.zshrc`, like you did for the `PYTHONPATH`, `PATH`... environment variables?**
  - **Can you add a page in the doc that is linked to the `CMakeList.txt` example and provide the standard procedure to build it and adapt the `CMakeList.txt` to other examples?** This could be very helpful to users like me who did not use `cmake` for a long time (but I think I am more the target of the Python version 😂)

c-joly, 2024-07-12:
Haha, I should be blind or maybe I was tired the day I wrote this issue... There is actually some explanations at the very end of the FAQ!  😅

However, it does not mention the possible problems that can occur if `cmake` cannot find the different packages. Can you please add a comment about the environment variable `CMAKE_PREFIX_PATH`?

costashatz, 2024-08-26:
@c-joly feel free to re-open the issue if #215 did not fix this :) Thanks again!

closed 2024-08-26 by costashatz