NOSALRO/robot_dart #204: Segfault on MacOS when shadows are on (GRAPHIC ON)

c-joly, 2024-06-28:
Hi @costashatz, here is in issue linked to the review https://github.com/openjournals/joss-reviews/issues/6771

Launching the examples in graphical mode on MacOS 14.5 (ARM64) results in a **segfault** due to OpenGL immediately (C++) and after about one second (Python bindings). **Deactivating the shadows solve the problem** (tested by updating some examples in Python and C++). 

I saw there is already a warning message mentioning unexpected behavior with shadows on MacOS. In turns out that with my configuration, there is no behavior at all (☠️). 

Do you think you can patch the examples to switch off the shadows by default, or better to deactivate them if the **Darwin** architecture (==MacOS) is detected? In Python, the test can be made easily thanks to the function `system` of the module `platform`.

costashatz, 2024-06-28:
@c-joly thank you for your hard work. I will try to reply to all of your issues.

> Deactivating the shadows solve the problem

Nice to hear!

> better to deactivate them if the **Darwin** architecture (==MacOS) is detected?

Yes. This is what I will do. I am already checking for **Darwin** architecture in my code. So this should be easy. It was just that shadows were working in some machines, and producing some small artifacts on others. But on yours, it seems that they do not work at all.

It's unclear to me why, but the easiest workaround is to indeed disable shadows by default on Mac.

Thanks again!

c-joly, 2024-06-28:
> It's unclear to me why, but the easiest workaround is to indeed disable shadows by default on Mac.

I found on some forum that there is no support anymore of OpenGL on MacOS. Some people mention that the bugs and compatibility issues may be worse in the future. It looks like Apple would like to force developers to use *Metal* on MacOS, which is now the only graphics API officially supported.

c-joly, 2024-07-04:
Hi @costashatz, a little update about these segfault. I figured out they happen only when my computer is docked and an extra screen is connected. I figured out by accident that I can run the examples in shadowed mode when I used my laptop without any extra screen... 🫤

costashatz, 2024-09-25:
This should be fixed by https://github.com/NOSALRO/robot_dart/commit/9412aa2936e6f331431708d2c4d04ee8af635463 (and see https://github.com/NOSALRO/robot_dart/issues/223). @c-joly let me know if I forgot something.

closed 2024-09-25 by costashatz