NOSALRO/robot_dart #205: Status bar not present on MacOS

c-joly, 2024-06-28:
Hi @costashatz, here is in issue linked to the review https://github.com/openjournals/joss-reviews/issues/6771

I found a minor bug when launching the examples with **GRAPHIC ON** on MacOS 14.5 (Arm64). The status bar, which should be at the bottom of the window is not present (completely black). It might be a bug due to the lack of support of `OpenGL` for MacOS (it works well on Ubuntu). 

If this bug cannot be solved easily, can you mention in the documentation that the status bar may not appear on MacOS? (Since the FAQ has a complete paragraph to describe what we can find in the status bar...)

costashatz, 2024-08-29:
This is connected to #204 and the HiDPI scaling issue. For now, I will add some information about this in the docs.

costashatz, 2024-09-25:
This should be fixed by https://github.com/NOSALRO/robot_dart/commit/9412aa2936e6f331431708d2c4d04ee8af635463 (and see https://github.com/NOSALRO/robot_dart/issues/223). @c-joly let me know if I forgot something.

closed 2024-09-25 by costashatz