NOSALRO/robot_dart #214: Work very good on WSL with Ubuntu 22.04.4 LTS

bstanciulescu, 2024-07-12:
Hi  @costashatz , this is a comment related to the JOSS review https://github.com/openjournals/joss-reviews/issues/6771
1. Successfull compilation and installation under the WSL/Ubuntu 22.04 layer on Windows 11 ! You should maybe add to the project's description the possibility of installing Robot_Dart under Windows 11/WSL.
1. First C++ tests successfully running; namely the `arm` and `arm_plain`.

bstanciulescu, 2024-07-12:
The previous example works also with the Python bindings.
However, there is a common bug in both C++ and Python graphics; the horizontal status bar is not displayed.

costashatz, 2024-09-03:
This is fixed via ab8c9c5a6a8ba3fe4d46ccb91ff79c969fcd06e5!

Thanks @bstanciulescu

closed 2024-09-03 by costashatz